chipsalliance/sv-tests #36: Set up travis

kgugala, 2019-08-27:
The CI should:

* Run all the tests
* Check PR's (code quality etc.)
* Push the result HTML to GH pages

tgorochowik, 2019-08-28:
Generating the report and checking code quality (for python scripts) should be up and running.

As mentioned in #39 - we need help of the project admins with setting up the github pages..
These need to be enabled in the repo settings and a github token variable needs to be added. (See: https://docs.travis-ci.com/user/deployment/pages/ )
CC: @mithro

mithro, 2019-08-28:
`GH_TOKEN` and `GH_USER`, `GIT_USER` and `GIT_EMAIL` should all be set on travis. `GH_TOKEN` will only appear on the master branch builds.

tgorochowik, 2019-08-30:
@mithro could you re-check the gh-pages configuration?

We used the variable you mentioned:
https://github.com/SymbiFlow/sv-tests/blob/817a83741f8c1860dd27e6062f770cc293fc8538/.travis.yml#L45

But the [build](https://travis-ci.com/SymbiFlow/sv-tests/jobs/229647215) fails with:
```
commit 817b54a3006bbd0c6188b0f72bd8a45fe39ae61f
Author: Deployment Bot (from Travis CI) <deploy@travis-ci.org>
Date:   Fri Aug 30 15:49:19 2019 +0000

    Deploy SymbiFlow/sv-tests to github.com/SymbiFlow/sv-tests.git:gh-pages

 index.thml | 59038 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 1 file changed, 59038 insertions(+)
Couldn't push the build to github.com/SymbiFlow/sv-tests.git:gh-pages

failed to deploy
```

mithro, 2019-09-02:
This looks like it is working now;
https://travis-ci.com/SymbiFlow/sv-tests/jobs/230163527
```
Deploying application
cd /tmp/d20190902-11568-1q1hzl8/work
Copying /home/travis/build/SymbiFlow/sv-tests/out/deploy contents to /tmp/d20190902-11568-1q1hzl8/work (workdir: /tmp/d20190902-11568-1q1hzl8/work)...
Configuring git committer to be Deployment Bot (from Travis CI) <deploy@travis-ci.org> (workdir: /tmp/d20190902-11568-1q1hzl8/work)
Preparing to deploy gh-pages branch to gh-pages (workdir: /tmp/d20190902-11568-1q1hzl8/work)
commit c9d547349a7635ec1a5d65cf67ce99bd1552a5a8
Author: Deployment Bot (from Travis CI) <deploy@travis-ci.org>
Date:   Mon Sep 2 20:08:10 2019 +0000
    Deploy SymbiFlow/sv-tests to github.com/SymbiFlow/sv-tests.git:gh-pages
 index.thml | 96238 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 1 file changed, 96238 insertions(+)
Doing the git push (workdir: /tmp/d20190902-11568-1q1hzl8/work)...
On branch gh-pages
Your branch is ahead of 'origin/gh-pages' by 1 commit.
  (use "git push" to publish your local commits)
nothing to commit, working tree clean
cd -
App has been pushed
Done. Your build exited with 0.
```
https://github.com/SymbiFlow/sv-tests/tree/gh-pages

mithro, 2019-09-02:
Looks like this is working with my addition of the `.nojekyll` file. See https://symbiflow.github.io/sv-tests/

PR #42 (2019-08-30): ci: deploy report using github pages
Uses variables from #36 

Closes #36

Depends on #41
commit b72fbf9b (2019-08-30): ci: deploy report using github pages

Closes #36
closed 2019-09-03 by mithro