chipsalliance/sv-tests #77: Investigate if outputting xUnit format makes sense

mithro, 2019-09-10:
If sv-tests could output xUnit XML, then the test results would be compatible with a wide range of tools. Could mean we can ditch creating our own HTML frontend?

> XML files in xUnit format can be consumed by a wide range of tools, such as build systems, IDEs and continuous integration servers.
>
> ## Schema
> 
> There are many schemas with minor differences.
> We use one that is compatible with Jenkins xUnit plugin, a copy is
> available under `tests/vendor/jenkins/xunit-plugin/junit-10.xsd` (see attached license).
> You may also find these resources useful:
> 
> - https://stackoverflow.com/questions/4922867/what-is-the-junit-xml-format-specification-that-hudson-supports
> - https://stackoverflow.com/questions/11241781/python-unittests-in-jenkins
> - [Jenkins (junit-10.xsd)](https://github.com/jenkinsci/xunit-plugin/blob/master/src/main/resources/org/jenkinsci/plugins/xunit/types/model/xsd/junit-10.xsd)
> - [JUnit-Schema (JUnit.xsd)](https://github.com/windyroad/JUnit-Schema/blob/master/JUnit.xsd)
> - [Windyroad (JUnit.xsd)](http://windyroad.com.au/dl/Open%20Source/JUnit.xsd)
> - [a gist (Jenkins xUnit test result schema)](https://gist.github.com/erikd/4192748)

piotr-binkowski, 2019-09-13:
I've checked the xUnit XML format and the format itself supports storing all the information that we use (runner log, test status, basic hierarchy - chapters, subchapters).

But the tools I found for visualizing the XML doesn't seem to produce results as readable as our current reports, especially when it comes to comparing different tools.
I tried [xunit-to-html](https://github.com/Zir0-93/xunit-to-html) as it was the most promising in my opinion and the results are shown below:
[image]
[image]
I prepared mock XML files using two approaches, first one groups test results by chapters and the other one groups them by runner.

I also checked the [xunit-viewer](https://github.com/lukejpreston/xunit-viewer) but the [demo](https://lukejpreston.github.io/xunit-viewer/) provided seemed like it also wouldn't suit our needs.

Maybe someone could suggest a more suitable tool for processing those XML reports

mithro, 2019-09-13:
You should just be able to feed your XML into xunit-viewer and see what it looks like?

mithro, 2019-09-13:
Maybe https://github.com/inorton/junit2html/blob/master/README.md ?

mithro, 2019-09-13:
Maybe https://github.com/extent-framework/extentreports-dotnet-cli

piotr-binkowski, 2019-09-16:
I also tried to use the `extentreports-dotnet-cli` tool but I couldn't get it to work on my PC, ie. running
> [pbinkowski@arch xunit]$ mono extent.exe -i /home/pbinkowski/projects/systemverilog/xunit/test.xml -o /home/pbinkowski/projects/systemverilog/xunit/ext/
> 
> extentreports-cli initializing ...
> Parsing test runner result file '/home/pbinkowski/projects/systemverilog/xunit/test.xml' ...
> The html report will be output to the folder '/home/pbinkowski/projects/systemverilog/xunit/ext/'.
> Report for '/home/pbinkowski/projects/systemverilog/xunit/test.xml' is complete.
> extentreports-cli finished.

would leave the output directory empty while at the same time log looks like everything went fine

piotr-binkowski, 2020-02-03:
I think we can close this since we ended up using a custom generated HTML + `.csv` report

closed 2020-02-03 by piotr-binkowski