chipsalliance/sv-tests #108: chapter-5/5.6.1--escaped-identifiers.sv

drom, 2019-09-17:
`sv-tests/tests/chapter-5/5.6.1--escaped-identifiers.sv` -- this test is passing by `tree-sitter-verilog` but reported as failing. Could you explain why?

piotr-binkowski, 2019-09-18:
I've checked the tests you listed and I discovered the following things:

- `12.7.5` fails because it is missing a `;` after `while(...)`, soon this will be fixed by #110
- `5.6.1` seems to fail because of missing `;` and it also fails when tested manually using `tree-sitter-cli` with error `5.6.1--escaped-identifiers.sv   1 ms    (MISSING ";" [8, 19] - [8, 19])`
- remaining tests from the `generic` subdirectory also seem to fail both in `tree-sitter-cli` with errors like `(MISSING simple_identifier [7, 9] - [7, 9])` and in our test suite

Are you sure that those tests passed on your side?

In general test is marked as failed in `tree-sitter-verilog` runner when any node has the [has_error](https://github.com/SymbiFlow/sv-tests/blob/master/tools/runners/tree_sitter_verilog.py#L13) property set to `True` and this has been the case in all mentioned tests.

tgorochowik, 2019-09-18:
> * `5.6.1` seems to fail because of missing `;` and it also fails when tested manually using `tree-sitter-cli` with error `5.6.1--escaped-identifiers.sv   1 ms    (MISSING ";" [8, 19] - [8, 19])`

Why is that error not seen in the report? This is a bug in the python runner and should be fixed in my opinion, otherwise the logs are pretty useless

piotr-binkowski, 2019-09-18:
Currently logs only show the part of the node hierarchy that has the `has_error` flag set, ie. in case of `5.6.1`:
> ...
(0, 0) source_file
(6, 0) module_declaration
(8, 2) module_or_generate_item
(8, 2) data_declaration
(8, 19) ;
...

and for a run that succeeded the log should be empty.
Maybe the runner can be enhanced to directly state at the end of the log that the error is in this particular line and what kind of error it is

drom, 2019-09-18:
@piotr-binkowski you are right. Cases are failing. It just the log was confusing.

drom, 2019-09-23:
I have released https://github.com/tree-sitter/tree-sitter-verilog v0.18.0 here that fixes these and many other issues. Please upgrade.

tgorochowik, 2019-10-03:
The misleading messages have been improved in #200 so I guess this can be closed now.

closed 2019-10-03 by tgorochowik