chipsalliance/sv-tests #109: Parser tests?

drom, 2019-09-17:
Does this test suite target SystemVerilog Parsers or Full Compilers?

Some of the tests failing on `tree-sitter-verilog` because it is bare parser, not a preprocessor, not a compiler.

For example:

## Preprocessor

Multiple Number tests: `/tests/generic/number/number_test_*.sv` include preprocessor directives like: `DIGITS`, `WIDTH`

Other examples:

```
sv-tests/tests/chapter-5/5.6.4--compiler-directives-begin-keywords.sv
sv-tests/tests/chapter-5/5.6.4--compiler-directives-pragma.sv
sv-tests/tests/chapter-5/5.6.4--compiler-directives-unconnected-drive.sv
...
```

## Reserved words

All `/tests/generated/keywords/5.6.2--keyword-*.sv` tests suppose to fail on reserved keywords used is an identifier. I would not expect that it is Parser's job to fail in this case. Is it?

## Scope failures

There are multiple tests for scope related issues that is expected to fail by the compiler. 

Example:

```
sv-tests/tests/generated/encapsulation/8.18--inherited_local_from_inside.sv
sv-tests/tests/generated/encapsulation/8.18--inherited_local_from_outside.sv
sv-tests/tests/generated/encapsulation/8.18--inherited_prot_from_outside.sv
sv-tests/tests/generated/encapsulation/8.18--local_from_outside.sv
sv-tests/tests/generated/encapsulation/8.18--prot_from_outside.sv
```

I don't think it is the job of a Parser to fail here.

mithro, 2019-09-17:
@drom The target is to be able to test compliance around all of parsers, synthesis and simulation. Maybe we need some type of tag which describes what is needed to pass a test?

drom, 2019-09-18:
On preprocessor issue:
For all bare parsers, input code should pass via the Verilog preprocessor.
We can use: `verilator -P -E <OPTIONS> mytop.sv > mytop.prep.sv`

You are right. For tests, that supposed to fail (like Reserved words, scope analysis, etc.) we should 
add the tag, describing at what stage compiler should fail.

Nic30, 2019-10-12:
Maybe we should also separate preprocessors from parsers/compilers entirely.

tgorochowik, 2019-11-04:
This is now resolved with #355, we can now specify a type for the tests and runners and then mix and match to select what can be run on what. Thanks for the report!

closed 2019-11-04 by tgorochowik