chipsalliance/sv-tests #122: Test quality improvement: Double-check tests that fail for all tools

hzeller, 2019-09-20:
Some test fail for all tools. This could mean they use too sophisticated SystemVerilog features. But more often than not, they contain information that is missing, such as this [number_test_23.sv](https://github.com/SymbiFlow/sv-tests/blob/master/tests/generic/number/number_test_23.sv) that is using macros without them being defined (that particular test came out of some other test-suite that just did syntax checking but all the tools are confused then because they can't resolve symbols; so here, we should define the macros).

Other tests contained other issues such discovered by @alainmarcel  and fixed in pull request #119 #116 #114 #113 #112 #111 (thanks @alainmarcel!).

So while still cranking out new tests, we should have a second look at the existing tests that fail.
I added some CSV output in the WIP pull request in https://github.com/SymbiFlow/sv-tests/commit/ce82b7964a5984dc57aea84d06e21c413706b3e8 ; with that, it is easy to look at these.

First, here is a fun overview of all the tests that are running successfully out of 496 (formulated as a SQL query for readability (and that is what I use with my personal tool for that), but `awk` or similar would work as well.)

```SQL
from csv:out/results.csv "select sum(Icarus), sum(Slang), sum(Sv2v_zachjs), sum(Verilator), sum(Yosys)"
+-------------+------------+------------------+----------------+------------+
| sum(Icarus) | sum(Slang) | sum(Sv2v_zachjs) | sum(Verilator) | sum(Yosys) |
+-------------+------------+------------------+----------------+------------+
|         122 |        198 |              202 |            183 |        117 |
+-------------+------------+------------------+----------------+------------+

from csv:out/results.csv "select 100.0*sum(Icarus)/count(*) as Ikaraus_percent, 100.0*sum(Slang)/count(*) as Slang_percent, 100.0*sum(Sv2v_zachjs)/count(*) as sv2v_percent, 100.0*sum(Verilator)/count(*) as verilator_percent, 100.0*sum(Yosys)/count(*) as yosys_percent"
+-----------------+---------------+--------------+-------------------+---------------+
| Ikaraus_percent | Slang_percent | sv2v_percent | verilator_percent | yosys_percent |
+-----------------+---------------+--------------+-------------------+---------------+
|         24.5968 |       39.9194 |      40.7258 |           36.8952 |       23.5887 |
+-----------------+---------------+--------------+-------------------+---------------+
```
Anyway, here is a list of all tests that fail for all the tested tools. Might be worthwhile taking a second look at these:
```SQL
from csv:out/results.csv select test where not Icarus and not Slang and not Sv2v_zachjs and not Verilator and not Yosys format "%s\n"
```

```
chapter-11/11.12--let_construct.sv
chapter-11/11.4.13--set_member.sv
chapter-11/11.4.14.4--dynamic_array_stream.sv
chapter-11/11.9--tagged_union.sv
chapter-11/11.9--tagged_union_member_access.sv
chapter-12/12.6.1--case_pattern.sv
chapter-12/12.6.1--casex_pattern.sv
chapter-12/12.6.1--casez_pattern.sv
chapter-12/12.6.2--if_pattern.sv
chapter-12/12.6.3--conditional_pattern.sv
chapter-12/12.7.5--dowhile.sv
chapter-5/5.10-structure-replication.sv
chapter-5/5.10-structures.sv
chapter-5/5.11-arrays-replication.sv
chapter-5/5.13-builtin-methods-arrays.sv
chapter-5/5.8-time-literals.sv
chapter-5/5.9-string-word-assignment.sv
chapter-8/8.13--inheritance.sv
chapter-8/8.14--override_member.sv
chapter-8/8.15--super.sv
chapter-8/8.20--virtual_method.sv
chapter-8/8.21--abstract_class.sv
chapter-8/8.22--dynamic_method_lookup.sv
chapter-8/8.23--scope_resolution.sv
chapter-8/8.24--out_of_block_methods.sv
chapter-8/8.25--parametrized_class_extend.sv
chapter-8/8.25.1--parametrized_class_scope_resolution.sv
chapter-8/8.26.2--implements.sv
chapter-8/8.26.2--implements_extends.sv
chapter-8/8.26.2--implements_multiple.sv
chapter-8/8.26.3--type_access_extends.sv
chapter-8/8.26.3--type_access_implements.sv
chapter-8/8.26.5--cast_between_interface_classes.sv
chapter-8/8.26.5--implemented_class_handle.sv
chapter-8/8.26.6.1--name_conflict_resolved.sv
chapter-8/8.26.6.2--parameter_type_conflict.sv
chapter-8/8.26.6.3--diamond_relationship.sv
chapter-8/8.26.7--partial_implementation.sv
chapter-8/8.27--forward_declaration.sv
chapter-8/8.5--parameters.sv
[... 159 lines cut ...]
generic/number/number_test_37.sv
generic/number/number_test_38.sv
generic/number/number_test_39.sv
generic/number/number_test_40.sv
generic/number/number_test_41.sv
generic/number/number_test_54.sv
generic/number/number_test_55.sv
generic/number/number_test_56.sv
generic/number/number_test_57.sv
generic/number/number_test_58.sv
generic/number/number_test_73.sv
generic/number/number_test_74.sv
generic/number/number_test_75.sv
generic/number/number_test_76.sv
generic/number/number_test_77.sv
generic/preproc/preproc_test_2.sv
generic/preproc/preproc_test_3.sv
generic/typedef/typedef_test_10.sv
generic/typedef/typedef_test_11.sv
generic/typedef/typedef_test_12.sv
generic/typedef/typedef_test_13.sv
generic/typedef/typedef_test_15.sv
generic/typedef/typedef_test_17.sv
generic/typedef/typedef_test_9.sv
```

hzeller, 2019-09-24:
Many (all?) of the generic tests can probably be safely removed as they were auto-generated from some separate source and might not be too helpful anyway.

The `chapter-*/` tests however should just be double-checked if they might contain typos, but these might just be legitimate failures across the board.

alainmarcel, 2019-09-24:
Are there any tests that are still auto generated? Is the make generate-tests generating tests? I have not been able to make that run. I’m only looking at the tests that are checked in git.

Thanks,
Alain

hzeller, 2019-09-24:
The tests in `tests/generic/` are the ones (and they are checked in to git). They were generated by some other set of test samples, but they were mostly just for syntax checking and many of them don't pass more scrutiny (e.g. the number_test_23.sv mentioned above that uses macros that were never defined).

alainmarcel, 2019-09-24:
I ran all tests with my tool, the list of failing tests in generic is small once you define the following macros on the tool command line:  +define+DIGITS=10 +define+WIDTH=2

 
ERROR  : tests/generic/iface/iface_class_test_6.sv
ERROR  : tests/generic/typedef/typedef_test_21.sv
ERROR  : tests/generic/typedef/typedef_test_17.sv
ERROR  : tests/generic/typedef/typedef_test_23.sv
ERROR  : tests/generic/typedef/typedef_test_15.sv
ERROR  : tests/generic/typedef/typedef_test_10.sv
ERROR  : tests/generic/member/class_member_test_16.sv
ERROR  : tests/generic/member/class_member_test_22.sv
ERROR  : tests/generic/member/class_member_test_3.sv
ERROR  : tests/generic/member/class_member_test_13.sv
ERROR  : tests/generic/member/class_member_test_28.sv
ERROR  : tests/generic/member/class_member_test_29.sv
ERROR  : tests/generic/preproc/preproc_test_11.sv
ERROR  : tests/generic/preproc/preproc_test_10.sv
ERROR  : tests/generic/preproc/preproc_test_13.sv
ERROR  : tests/generic/preproc/preproc_test_9.sv
ERROR  : tests/generic/preproc/preproc_test_12.sv
ERROR  : tests/generic/class/class_test_35.sv
ERROR  : tests/generic/class/class_test_55.sv

Each test only require small fixing to pass, I'll submit all the the changes.These tests are still of value, for instance the "number" tests make good testing of the preprocessor and Lexer.

alainmarcel, 2019-09-24:
All of these new tests are also incorrect, wrap the string function call in a function to make them correct tests:
module top();
   function toto();      
    string a;
    a.octtoa(12);
   endfunction   
endmodule

ERROR  : tests/chapter-6/6.16.13--string_octtoa.svERROR  : tests/chapter-6/6.16.11--string_itoa.sv
ERROR  : tests/chapter-6/6.16.12--string_hextoa.sv
ERROR  : tests/chapter-6/6.16.14--string_bintoa.sv
ERROR  : tests/chapter-6/6.16.15--string_realtoa.sv

-- Alain Marcel 

    On Tuesday, September 24, 2019, 4:30:44 PM PDT, Alain Dargelas <alainmarcel@yahoo.com> wrote:  
 
 I ran all tests with my tool, the list of failing tests in generic is small once you define the following macros on the tool command line:  +define+DIGITS=10 +define+WIDTH=2

 
ERROR  : tests/generic/iface/iface_class_test_6.sv
ERROR  : tests/generic/typedef/typedef_test_21.sv
ERROR  : tests/generic/typedef/typedef_test_17.sv
ERROR  : tests/generic/typedef/typedef_test_23.sv
ERROR  : tests/generic/typedef/typedef_test_15.sv
ERROR  : tests/generic/typedef/typedef_test_10.sv
ERROR  : tests/generic/member/class_member_test_16.sv
ERROR  : tests/generic/member/class_member_test_22.sv
ERROR  : tests/generic/member/class_member_test_3.sv
ERROR  : tests/generic/member/class_member_test_13.sv
ERROR  : tests/generic/member/class_member_test_28.sv
ERROR  : tests/generic/member/class_member_test_29.sv
ERROR  : tests/generic/preproc/preproc_test_11.sv
ERROR  : tests/generic/preproc/preproc_test_10.sv
ERROR  : tests/generic/preproc/preproc_test_13.sv
ERROR  : tests/generic/preproc/preproc_test_9.sv
ERROR  : tests/generic/preproc/preproc_test_12.sv
ERROR  : tests/generic/class/class_test_35.sv
ERROR  : tests/generic/class/class_test_55.sv

Each test only require small fixing to pass, I'll submit all the the changes.These tests are still of value, for instance the "number" tests make good testing of the preprocessor and Lexer.

mithro, 2019-09-25:
BTW @alainmarcel Thank you for all your efforts! It's greatly appreciated.

hzeller, 2019-09-25:
Related: pull request #140 by @alainmarcel which affects one of these files.

Nic30, 2019-10-12:
#314 tests/generic/member/class_member_test_13.sv

piotr-binkowski, 2020-02-03:
I think that we can close this as this was handled in #199 and other PRs

closed 2020-02-03 by piotr-binkowski