chipsalliance/sv-tests #138: Test failures when using Icarus Verilog

martinwhitaker, 2019-09-23:
I had a quick look at the test failures when using Icarus Verilog. I spotted two recurring problems:

1. Many tests do not include a top-level module. iverilog returns an error in this case, as there is nothing to elaborate. I have just enhanced the iverilog '-i' option (ignore missing modules) to also suppress this error, so using that option will fix many failures. Of course the test code will then only be checked for syntax errors.

2. A number of tests contain an 'always' construct with no waits or delays. iverilog returns an error in this case, because it will cause the simulator to go into an infinite loop. It would be best not to code tests like that.

I did spot quite a few broken tests, e.g. number_test_63.v uses invalid syntax.

mithro, 2019-09-23:
@martinwhitaker - The test suite is only very new, so if you do fine issues please log bugs.

tgorochowik, 2020-02-03:
The `-i` switch has been added a long time ago, we need to take a look at the tests which use `always` to determine if that is still a problem.

piotr-binkowski, 2020-02-07:
I've checked the tests and there are some that have this `always` infinite loop issue, I've created a small PR #622 to fix those by adding a sensitivity list to `always` blocks.

piotr-binkowski, 2020-02-21:
#622 was merged so the issue with `always` blocks should now be fixed and I think we can close this

closed 2020-02-21 by piotr-binkowski