chipsalliance/sv-tests #258: dependabot ignores tree-sitter-verilog changes

drom, 2019-10-07:
Somehow dependabot ignores tree-sitter-verilog changes for a week. Do I need to do something special on the `tree-sitter-verilog` repo to see updates faster?

tgorochowik, 2019-10-07:
#189 seems to include most of the recent changes (excluding the ones made yesterday). The changes from yesterday are excluded because of #230 - as you can see we deliberately lowered the frequency at which dependabot operates as we have many submodules and it made the CI too occupied for us to test our actual changes.

If you want the submodule with `tree-sitter-verilog` to be updated more frequently feel free to create the PR with bumps yourself (or commenting on the ones created by dependabot) - we will try to review such PRs with a higher priority.

drom, 2019-10-07:
Thank you @tgorochowik . That makes sense!

closed 2019-10-07 by drom