chipsalliance/sv-tests #261: Add 'make format' for format Python code

hzeller, 2019-10-07:
The CI catches these, but would be good if we can have that beforehand.

hzeller, 2019-10-08:
When running 'make format', this complains.
Is this because of actual formatting complaints or because my laptop has Python 2.7 installed?

```bash
$ make format
flake8 generators/fx68k generators/ibex generators/path_generator generators/swerv generators/template_generator generators/yosys_hana tools/BaseRunner.py tools/check-runners tools/runner tools/sv-report tools/BaseRunner.py tools/runners/Slang.py tools/runners/Yosys.py tools/runners/tree_sitter_verilog.py tools/runners/Icarus.py tools/runners/Verilator.py tools/runners/Odin.py tools/runners/Sv2v_zachjs.py
generators/path_generator:49:1: E999 SyntaxError: only named arguments may follow *expression
tools/runner:113:9: F821 undefined name 'PermissionError'
tools/runner:113:26: F821 undefined name 'FileExistsError'
tools/runners/tree_sitter_verilog.py:28:1: E999 SyntaxError: only named arguments may follow *expression
```

mithro, 2019-10-08:
This is what prjxray does for `make format`. 

See
https://github.com/SymbiFlow/prjxray/blob/daa516585eb31fe7f05cdb8d4f080d4cb96751d1/Makefile#L54-L77

```Makefile
# Auto formatting of code.
# ------------------------
FORMAT_EXCLUDE = $(foreach x,$(ALL_EXCLUDE),-and -not -path './$(x)/*')

CLANG_FORMAT ?= clang-format-5.0
format-cpp:
	find . -name \*.cc $(FORMAT_EXCLUDE) -print0 | xargs -0 -P $$(nproc) ${CLANG_FORMAT} -style=file -i
	find . -name \*.h $(FORMAT_EXCLUDE) -print0 | xargs -0 -P $$(nproc) ${CLANG_FORMAT} -style=file -i

format-docs:
	./.github/update-contributing.py

PYTHON_FORMAT ?= yapf
format-py:
	$(IN_ENV) find . -name \*.py $(FORMAT_EXCLUDE) -print0 | xargs -0 -P $$(nproc) yapf -p -i

TCL_FORMAT ?= utils//tcl-reformat.sh
format-tcl:
	find . -name \*.tcl $(FORMAT_EXCLUDE) -print0 | xargs -0 -P $$(nproc) -n 1 $(TCL_FORMAT)

format: format-cpp format-docs format-py format-tcl
	@true

.PHONY: format format-cpp format-py format-tcl
```

tgorochowik, 2019-10-08:
@hzeller it seems that your flake8 instance is installed for python2, not python3, we will have to check if we can be smarter about this and require it to be installed for py3 (I guess we should be able to invoke it differently)

PR #263 (2019-10-08): makefile: ci: use make format for pep8 checks
This adds `make format` target for pep8 checks and replaces check in CI with it
Fixes #261
PR #277 (2019-10-10): Modify python format checks to use yapf
This PR modifies `make format` command to use `yapf` to automatically adjust python code formatting
and adjusts CI to check if the code is formatted properly.
Closes #261
closed 2019-10-10 by tgorochowik