chipsalliance/sv-tests #281: run preprocessor before tree-sitter-verilog

drom, 2019-10-09:
`tree-sitter-verilog` is pure SystemVerilog parser and doesn't include preprocessor. Many tests in the suite include tick-defines and need to be preprocessed before passing to `tree-sitter-verilog`. One simple way of solving this issue is to pass every source file via verilator preprocessor before passing it to tree-sitter-verilog like that:

```
verilator -E -P source.sv > source.prep.sv
```

tgorochowik, 2019-10-10:
I guess we could add that as a workaround, but I don't think it is a perfect solution. If there is a bug in the preprocessor in verilator, it would still mark the test as failed for tree-sitter-verilog.

I think we should go with tagging the tests and tools as proposed in #109 instead, and just not run all tests on everything.

CC @hzeller

drom, 2019-10-11:
Does all the rest of the tools have own preprocessor?

tgorochowik, 2019-11-04:
We are certainly going to be working with more tools that are parsers-only, therefore as mentioned in #355, with #109 we can now specify the types for test and runners. Thanks for the report!

closed 2019-11-04 by tgorochowik