chipsalliance/sv-tests #314: tests/generic/member/class_member_test_13.sv missing define

Nic30, 2019-10-12:
https://github.com/SymbiFlow/sv-tests/blob/master/tests/generic/member/class_member_test_13.sv#L9

I am seeing quite lot of missing defines in preprocessor, like in this file.
Is just defined somewhere else or it is really missing?

alainmarcel, 2019-10-12:
I think they are just missing, or part of a make system that would pass them as command line +define+ options to the ore processor. The tools should give a nice error when a define is missing.

Nic30, 2019-10-12:
So do we want to
* add definition
* or remove macro entirely
* or mark it as should fail
* or pass the definition from the runner?

alainmarcel, 2019-10-12:
Please add a simple definition in the file itself for all tests. 
There should be just one test that makes sure the tools support one way of specifying macros from the command line.

mithro, 2019-10-12:
These tests came from another test suite and is seems the defines where lost in the process.

tgorochowik, 2019-10-14:
Let's just remove remove the defined symbol usage here. This is tested in other tests anyway. See #320

PR #320 (2019-10-14): Fix generic class test 13
Fixes #314
commit 5a1ca583 (2019-10-14): Fix generic class test 13

Fixes #314
closed 2019-10-14 by piotr-binkowski