chipsalliance/sv-tests #315: SweRV RISC-V: incomplete headers, no header guards + wrong compile order

Nic30, 2019-10-12:
Hello,

SweRV RISC-V files do have problems mentioned in topic.
The code is working actually, but correct compile order has to be specified.
Namely preprocessor can be called only for a specific files and not for all files like it is happening now as all files are in file list.

I am not sure if tools should try to solve this problem, or the test is wrong. What do you think?

hzeller, 2020-12-18:
And currently a file is referenced that can't be opened - which all tools complain about in their respective ways (third_party/cores/swerv/configs/snapshots/default/common_defines.vh).

Time for #998 ...

joannabrozek, 2021-02-16:
Fixed in #1300, #1341 and #1344.

closed 2021-02-16 by joannabrozek