chipsalliance/sv-tests #323: Incorrect tests: forward typedefs

MikePopoloski, 2019-10-15:
Forwarding typedefs are required to resolve to a valid type declaration (see 6.18). Tests that I noticed have this problem:
enum_test_0
struct_test_0
union_test_0

You might just want to make these expect to fail; other tools incorrectly allow this.

wsnyder, 2020-01-10:
I believe these are legal tests. What IEEE means is if the type is ever used it must resolve to a type.

Having forward decls that never resolve is common in real designs for the same reason that "class Foo;" in C++ often never defines or uses Foo - it's in some header for an implementation that isn't needed by this compilation due to ifdef or other reasons.

MikePopoloski, 2020-01-10:
I can't say anything about what the spec writers *meant* when they wrote it but that interpretation is not what the spec actually says. Furthermore, other tools seem to agree. Here is VCS:
```
Error-[SV-UFTD] Undefined forward typedef
testbench.sv, 6
  The definition for the forward typedef 'C' was not found in the scope where 
  it was declared.
  Provide the definition of the forward typedef.
```
Cadence:
```ncvlog: *E,TYDMSC (testbench.sv,6|10): The class specified in the typedef was never declared.```
RivieraPro:
```ERROR VCP5263 "Unresolved forward typedef C definition." "testbench.sv" 6  13```

MikePopoloski, 2021-03-27:
Fixed a while back.

closed 2021-03-27 by MikePopoloski