chipsalliance/sv-tests #339: Add moore to test suite

fabianschuiki, 2019-10-18:
I absolutely love your compliance test suite. It's an absolutely excellent and necessary idea! Is there any chance to add the [moore] compiler to it? Any guidance on how to contribute would be awesome!

[moore]: http://llhd.io/

mithro, 2019-10-18:
If the moore compiler is open source, then we can definitely add it!

The first step would be to get your compiler into our conda packages at https://github.com/SymbiFlow/conda-packages

Then follow the examples in https://github.com/SymbiFlow/sv-tests/pull/316 or https://github.com/SymbiFlow/sv-tests/pull/249

fabianschuiki, 2019-10-18:
> If the moore compiler is open source, then we can definitely add it!

It is! I'll add a corresponding package to `conda-packages` and the necessary additions to this repo in a PR.

fabianschuiki, 2019-10-24:
Will be done once #340 lands.

PR #340 (2019-10-31): Add moore compiler
Add the [moore] compiler to the suite of tools. Fixes #339.

[moore]: https://github.com/fabianschuiki/moore
closed 2019-10-31 by tgorochowik