chipsalliance/sv-tests #354: Broken tests imported from Yosys

zachjs, 2019-10-21:
The following tests reference an undefined macro `outfile`:
* `yosys/tests/asicworld/code_hdl_models_arbiter_tb.sv`
* `yosys/tests/asicworld/code_verilog_tutorial_counter_tb.sv`
* `yosys/tests/asicworld/code_verilog_tutorial_first_counter_tb.sv`
* `yosys/tests/asicworld/code_verilog_tutorial_fsm_full_tb.sv`

The test `yosys/tests/errors/syntax_err04.v` seems to expect `wire [3]x;` to be syntactically invalid, but it  is perfectly valid as far as I am aware.

The test `yosys/tests/simple/memory.v` uses `bit` as the name of a port. This could be fixed using `begin_keywords`.

tgorochowik, 2020-02-03:
Thanks for the report (and sorry for the delay).

@piotr-binkowski could we verify that and maybe blacklist those tests if needed?

piotr-binkowski, 2020-02-06:
I've looked through the mentioned tests and `asicworld` and `memory.v` got blacklisted some time ago.

Regarding the `syntax_err04.v` the SystemVerilog spec says that packed arrays can be made of only single bit data types (7.4.1) and `wire` is a net type so it seems that this test is valid.

closed 2020-02-10 by zachjs