chipsalliance/sv-tests #427: A broken basejump_stl submodule causes our travis to fail

tgorochowik, 2019-11-12:
See: https://github.com/bespoke-silicon-group/basejump_stl/issues/187

There is also some discussion in #426.
Generally we do not want to disable fetching the submodules completely.

We should either find a way to select the submodules to clone by hand (probably not a good idea considering long-term), or allow travis to continue if some modules fail to clone (although this should probably be somehow indicated in the report - we don't want this to happen quietly).

hzeller, 2019-11-17:
Relevant pull-request to watch: https://github.com/bespoke-silicon-group/basejump_stl/pull/188

mithro, 2020-01-10:
This seems to be fixed now.

tgorochowik, 2020-02-03:
It is, both with a workaround in our repo and in mainline basejump.

closed 2020-02-03 by tgorochowik