chipsalliance/sv-tests #437: Runner failures from "codec can't decode byte"

wsnyder, 2019-11-16:
I tried the tests for the first time on a Ubuntu 18.04 LTS system and I'm getting UNICODE failure on some tests.

    python3 --version
    Python 3.6.8
    make tests RUNNERS=Verilator -j 5 -k
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--inherited_local_from_inside.sv --out ./out//logs/Verilator/generated/encapsulation/8.18--inherited_local_from_inside.sv.log --quiet
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--prot_from_inside.sv --out ./out//logs/Verilator/generated/encapsulation/8.1$
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--inherited_local_from_outside.sv --out ./out//logs/Verilator/generated/encap$
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--inherited_prot_from_inside.sv --out ./out//logs/Verilator/generated/encapsu$
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--local_from_inside.sv --out ./out//logs/Verilator/generated/encapsulation/8.$
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--inherited_prot_from_outside.sv --out ./out//logs/Verilator/generated/encaps$
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--local_from_outside.sv --out ./out//logs/Verilator/generated/encapsulation/8$
    ./tools/runner --runner Verilator --test generated/encapsulation/8.18--prot_from_outside.sv --out ./out//logs/Verilator/generated/encapsulation/8.$
    ./tools/runner --runner Verilator --test chapter-5/5.7.1--integers-signed.sv --out ./out//logs/Verilator/chapter-5/5.7.1--integers-signed.sv.log -$
    ERROR   | Unable to parse test file: 'ascii' codec can't decode byte 0xe2 in position 295: ordinal not in range(128)
    Makefile:94: recipe for target 'out//logs/Verilator/chapter-5/5.7.1--integers-signed.sv.log' failed
    make: *** [out//logs/Verilator/chapter-5/5.7.1--integers-signed.sv.log] Error 1
    make: Target 'tests' not remade because of errors.

This seems because e.g. third_party/cores/ariane/include/ariane_axi_pkg.sv has unicode in it.  A solution might be suggested here:
   https://stackoverflow.com/questions/9942594/unicodeencodeerror-ascii-codec-cant-encode-character-u-xa0-in-position-20

Also another minor thing, if I run "make test" twice I get the 8.18 tests printed again (see above). Looks like these aren't making output logfiles which seems suspicious.

piotr-binkowski, 2020-01-29:
I think that this issue was fixed by cad8b5a
And regarding 8.18 tests, I created a PR for that.

tgorochowik, 2020-02-03:
Which PR @piotr-binkowski ? Is this already resolved?

piotr-binkowski, 2020-02-03:
It's #594 and it is already resolved

closed 2020-02-03 by piotr-binkowski