chipsalliance/sv-tests #438: "vmain not found" false-fails ~100 Verilator tests

wsnyder, 2019-11-16:
There's a "vmain" harness problem that seems to seriously penalize Verilator:

>       out/logs/Verilator/chapter-20/20.10--error.sv.log
>       out/logs/Verilator/chapter-20/20.10--fatal.sv.log
>       out/logs/Verilator/chapter-20/20.10--info.sv.log
>       out/logs/Verilator/chapter-12/12.8--break.sv.log
>       out/logs/Verilator/chapter-12/12.8--continue.sv.log
>       out/logs/Verilator/chapter-13/13.3--task.sv.log
>       out/logs/Verilator/chapter-13/13.4--function.sv.log
>       out/logs/Verilator/chapter-13/13.4.1--function-return-assignment.sv.log
>       out/logs/Verilator/chapter-13/13.4.1--function-return.sv.log
>       out/logs/Verilator/chapter-13/13.4.2--function-automatic.sv.log
>       out/logs/Verilator/chapter-20/20.7--array-queries-multi-dim.sv.log
>       out/logs/Verilator/chapter-20/20.15--random.sv.log
>       out/logs/Verilator/chapter-20/20.5--itor.sv.log
>       out/logs/Verilator/chapter-20/20.5--real-bits-conv.sv.log
>       out/logs/Verilator/chapter-20/20.5--rtoi.sv.log
>       out/logs/Verilator/chapter-20/20.6--bits.sv.log
>       out/logs/Verilator/chapter-20/20.6--bits_type.sv.log
>       out/logs/Verilator/chapter-20/20.7--array-queries.sv.log
>       out/logs/Verilator/chapter-20/20.8--atan2.sv.log
>       out/logs/Verilator/chapter-20/20.8--ceil.sv.log
>       out/logs/Verilator/chapter-20/20.8--clog2.sv.log
>       out/logs/Verilator/chapter-20/20.8--exp.sv.log
>       out/logs/Verilator/chapter-20/20.8--floor.sv.log
>       out/logs/Verilator/chapter-20/20.8--hypot.sv.log
>       out/logs/Verilator/chapter-20/20.8--ln.sv.log
>       out/logs/Verilator/chapter-20/20.8--log10.sv.log
>       out/logs/Verilator/chapter-20/20.8--pow.sv.log
>       out/logs/Verilator/chapter-20/20.8--sqrt.sv.log
>       out/logs/Verilator/chapter-20/20.9--isunknown.sv.log
>       out/logs/Verilator/chapter-20/20.9--onehot.sv.log
>       out/logs/Verilator/chapter-20/20.9--onehot0.sv.log
>       out/logs/Verilator/chapter-13/13.4.3--const-function.sv.log
>       out/logs/Verilator/chapter-20/20.10--warning.sv.log
>       out/logs/Verilator/chapter-10/10.4.1--blocking-assignment.sv.log
>       out/logs/Verilator/chapter-11/11.10.1--string_compare.sv.log
>       out/logs/Verilator/chapter-11/11.10.1--string_copy.sv.log
>       out/logs/Verilator/chapter-11/11.4.12.2--string_concat_op.sv.log
>       out/logs/Verilator/chapter-11/11.4.12.2--string_repl_op.sv.log
>       out/logs/Verilator/chapter-11/11.4.14.3--unpack_stream_inv.sv.log
>       out/logs/Verilator/chapter-21/21.3--feof.sv.log
>       out/logs/Verilator/chapter-21/21.3--fgetc.sv.log
>       out/logs/Verilator/chapter-21/21.3--fgets.sv.log
>       out/logs/Verilator/chapter-21/21.3--file.sv.log
>       out/logs/Verilator/chapter-21/21.3--fread.sv.log
>       out/logs/Verilator/chapter-21/21.3--fscanf.sv.log
>       out/logs/Verilator/chapter-21/21.3--sscanf.sv.log
>       out/logs/Verilator/chapter-21/21.6--test.sv.log
>       out/logs/Verilator/chapter-21/21.6--value.sv.log
>       out/logs/Verilator/chapter-26/26.3--package-ref.sv.log
>       out/logs/Verilator/chapter-6/6.13--void.sv.log
>       out/logs/Verilator/chapter-6/6.19--enum_xx_inv.sv.log
>       out/logs/Verilator/chapter-6/6.19--enum_xx_inv_order.sv.log
>       out/logs/Verilator/chapter-6/6.19.3--enum_type_checking_inv.sv.log
>       out/logs/Verilator/chapter-6/6.19.4--enum_numerical_expr_no_cast.sv.log
>       out/logs/Verilator/chapter-7/arrays/multidimensional/copy.sv.log
>       out/logs/Verilator/chapter-7/arrays/multidimensional/subarrays.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/equality.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/onebit.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/operations.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/dimensions.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/high.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/increment.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/left.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/low.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/right.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/size.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/querying-functions/unpacked-dimensions.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/slice-equality.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/slice.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/treat-as-integer.sv.log
>       out/logs/Verilator/chapter-7/arrays/packed/variable-slice-zero.sv.log
>       out/logs/Verilator/chapter-7/arrays/unpacked/assignments.sv.log
>       out/logs/Verilator/chapter-7/arrays/unpacked/equality.sv.log
>       out/logs/Verilator/chapter-7/arrays/unpacked/onebit.sv.log
>       out/logs/Verilator/chapter-7/memories/read-write.sv.log
>       out/logs/Verilator/chapter-7/unions/packed/basic.sv.log
>       out/logs/Verilator/generated/sampled_functions/20.13--past.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--acos.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--acosh.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--asin.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--asinh.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--atan.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--atanh.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--cos.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--cosh.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--sin.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--sinh.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--tan.sv.log
>       out/logs/Verilator/generated/trig_functions/20.8--tanh.sv.log

These tests above, e.g. 11.10.1--string_copy.sv are reported as fails. When I run this in Verilator the test passes.  This seems to be due to a harness problem I see in the log:

    scr.sh: 3: scr.sh: ./vbuild/vmain: not found

>       out/logs/Verilator/chapter-5/5.10-structure-arrays-illegal.sv.log
>       out/logs/Verilator/chapter-6/6.12--real_bit_select.sv.log
>       out/logs/Verilator/chapter-6/6.12--real_bit_select_idx.sv.log
>       out/logs/Verilator/chapter-6/6.20.5--specparam_inv.sv.log
>       out/logs/Verilator/chapter-13/13.4.1--function-void-return.sv.log
>       out/logs/Verilator/chapter-6/6.5--variable_mixed_assignments.sv.log
>       out/logs/Verilator/chapter-6/6.5--variable_multiple_assignments.sv.log
>       out/logs/Verilator/chapter-6/6.5--variable_redeclare.sv.log

For these tests, Verilator returns an error, and the test says it should_fail, so Verilator is correct, but verilator is reported in "red" for the test.  Suspect related to above.

wsnyder, 2019-12-03:
As this issue is blocking a huge number of tests some assistance in resolving this would be appreciated..

Pull request #470 made an attempt at this, but keeps hitting Travis issues.

mithro, 2019-12-03:
@wsnyder - FYI

I landed some improvements to Kokoro yesterday, please rebase changes onto master if you are having issues there.

I'm working on trying to get Travis working better via splitting the tools apart at https://github.com/SymbiFlow/sv-tests/pull/471 -- hoping to land that sometime today / tomorrow.

Make sure you rebase onto the latest master.

wsnyder, 2019-12-04:
Should be ready for merge.

wsnyder, 2019-12-05:
Pull request merged.

closed 2019-12-05 by wsnyder