chipsalliance/sv-tests #444: Suggested README updates

wsnyder, 2019-11-16:
Please consider improving the README with:

- How to use RUNNERS= to run just one runner - I figured it out but would be nice to document it.

- How to run one test, e.g.

    ./tools/runner --runner Verilator --test tests/chapter-11/11.10.1--string_copy.sv

(it would be nice if --out could be assumed from the --test name, so didn't show that above.)

towoe, 2020-01-31:
I agree that some updates would be good.

Maybe also list the requirements, e.g. some instructions how to fulfill those.
```
pip3 install -U --user -r conf/requirements.txt
```

Kakadu, 2025-08-09:
It looks like mentioned command no longer work ((

````
 ✗ ./tools/runner --runner Verilator --test tests/chapter-11/11.10.1--string_copy.sv -o out
ERROR   | Unable to load runner module: No module named 'Verilator'    

````