chipsalliance/sv-tests #451: Test failing on kokoro

mithro, 2019-11-20:
https://source.cloud.google.com/results/invocations/8732fac4-33ea-465d-80fc-9b7eee097a5e/log

```
./tools/runner --runner Verilator --test generated/integers/6.11--integer_unsigned_time.sv --out ./out//logs/Verilator/generated/integers/6.11--integer_unsigned_time.sv.log --quiet
./tools/runner --runner Verilator --test generated/integers/6.11--integer_signed_shortint.sv --out ./out//logs/Verilator/generated/integers/6.11--integer_signed_shortint.sv.log --quiet
make: fork: Cannot allocate memory

[ID: 9102836] Build finished after 798 secs, exit value: 1

Warning: Permanently added 'localhost' (ECDSA) to the list of known hosts.
[00:44:09] Collecting build artifacts from build VM
Build script failed with exit code: 1
[00:44:58] Archiving artifacts
[00:45:08] Running blaze commands.
```

mithro, 2019-11-20:
@hzeller Any idea what is happening here?

mithro, 2019-11-20:
```
========================================
Host CPU
----------------------------------------
Cores: 4

Memory
----------------------------------------
MemTotal:       15400292 kB
MemFree:        13592056 kB
MemAvailable:   14853488 kB
```

mithro, 2019-11-20:
`+ make tests USE_ALL_RUNNERS=1 -j4`

mithro, 2019-11-20:
I wonder if there should be something restricting the memory usage of each tool run?

mithro, 2019-11-20:
`The getrlimit() and setrlimit() system calls get and set resource limits.  Each resource has an associated soft and hard limit, as defined by the rlimit structure:`

mithro, 2019-11-20:
> You could set the hard resource limit (setrlimit for RLIMIT_AS resource) before execve(). The program will not be able to allocate more than that amount of memory. If it tries to do so, memory allocation calls (brk, mmap, mremap) will fail.

hzeller, 2019-11-20:
I think at this point it is only doing timeouts, but memory can be excessive in particular when bugs in tools are hit. We discussed rlimit, but I think the only limit currently implemented was timeout.

mithro, 2019-11-20:
It looks like it is already doing memory limit?
https://github.com/SymbiFlow/sv-tests/blob/cfffbcdf7a9b6dd04672f2e070febf12082aa968/tools/BaseRunner.py#L9-L11

mithro, 2019-11-20:
The script also seems to do a `+ make tests USE_ALL_RUNNERS=1 -j4` -- so I would expect 4gb * 4 processes == 16g?

hzeller, 2019-11-20:
Ah yes, now I remember: I implemented that memory limit myself :)

So maybe available 12G are then not enough. In particular as the memory hogging programs will stick around for a while it means that at some point we have collected 4 running processes, all super large. Slang and Verilator in particular have a huge memory need, but all the others seem to be fine.

Maybe we should cap the memory at 3G or run will less processes.

mithro, 2019-11-20:
Actually, it is unclear to me if the rlimit applies to all children or *each* child can get 4gb (IE if it launches a second child it can take 8gb?).

I wonder if we should be using cgroups or similar on Kokoro? Something like https://wiki.archlinux.org/index.php/cgroups#Matlab ?

mithro, 2019-11-20:
```       cgexec -g *:test1 ls
       runs command ls in control group test1 in all mounted controllers.

       cgexec -g cpu,memory:test1 ls -l
       runs command ls -l in control group test1 in controllers cpu and memory.

       cgexec -g cpu,memory:test1 -g swap:test2 ls -l
       runs command ls -l in control group test1 in controllers cpu and memory and control group test2 in controller swap.
```

mithro, 2019-11-20:
```
sudo cgcreate -a $(whoami) -t $(whoami) -g memory,cpu:sv-tests
cgcreate -g memory,cpu:sv-tests/Verilator
```

mithro, 2019-11-20:
I added using cgroups on kokoro at https://github.com/SymbiFlow/sv-tests/pull/452/commits/5ca87f7600879b30867f3f87e062d75a3914ec2b

mithro, 2019-11-20:
Will see if it fails in https://github.com/SymbiFlow/sv-tests/pull/452

mithro, 2020-01-10:
It seems to be working at the moment?

tgorochowik, 2020-02-03:
Kokoro seems to work fine, let's close this.

closed 2020-02-03 by tgorochowik