chipsalliance/sv-tests #457: Bumping dependencies

zachjs, 2019-11-21:
I'm interested in bumping the version of sv2v tested in the report. From the logs, it has pulled down `zachjs-sv2v-3979d294` (built over a month ago), despite much newer versions being available at https://anaconda.org/SymbiFlow/zachjs-sv2v/files. How can I bump the version under this setup?

tgorochowik, 2019-11-21:
@mithro is that possible that the reason is that the commit hash is used as the version, and conda sorts the packages by that and doesn't recognize what is most up to date?

tgorochowik, 2019-11-21:
I appears so judging by the download count of various packages, we should fix that ASAP (and somehow get rid of the older packages so they don't get sorted as newer that the actual versions)

tgorochowik, 2019-11-22:
@zachjs it should be now fixed with https://github.com/SymbiFlow/conda-packages/pull/60 merged.

Thanks for the report!

closed 2019-11-22 by tgorochowik