chipsalliance/sv-tests #458: Problem with the CI build's slang binary?

MikePopoloski, 2019-11-21:
I noticed that the recently updated slang build fails on literally every test. I've run the same version locally and don't observe that failure... is the binary actually being built and run properly in the CI build?

tgorochowik, 2019-11-21:
It seems that the current binary from conda is indeed broken:
https://anaconda.org/SymbiFlow/slang/308467aba44b3456b5106acc1d4877c47e8bc5b6/download/linux-64/slang-308467aba44b3456b5106acc1d4877c47e8bc5b6-20191120_154749.tar.bz2

Running it fails:
```
./slang-driver sv-tests/tests/chapter-5/5.10-structure-arrays.sv --ast-json -
Illegal instruction (core dumped)
```

tgorochowik, 2019-11-22:
The older conda packages have been removed because of https://github.com/SymbiFlow/conda-packages/issues/59, so the links no longer works, but I tested them yesterday and judging by the package dates it seems that it started happening after https://github.com/SymbiFlow/conda-packages/pull/52 - however the release build works just fine for me when I build it locally, so it has to be something related to the versions of the tools used in travis.

tgorochowik, 2019-11-22:
It could also be triggered by the threading changes (https://github.com/MikePopoloski/slang/commit/c0cd916fcafe2a9acc047baef07d1561ea620358 and https://github.com/MikePopoloski/slang/pull/272) A question arises: why was that fix needed in Travis (Ubuntu), but worked fine on other systems - is that because of the version of GCC?

MikePopoloski, 2019-11-22:
Yeah, perhaps whatever version of GCC is being used by Travis is old and buggy?

MikePopoloski, 2019-12-03:
Any chance you guys will be fixing this? It seems a shame to say you include slang in your report when in reality the binary you're testing with doesn't even run.

hzeller, 2019-12-03:
@mithro what needs to be done to get the conda package for slang in https://github.com/SymbiFlow/conda-packages up to generate a running binary ?
We should probably make sure to have a somewhat recent compiler.

mithro, 2019-12-03:
It's building with GCC-8 which is pretty recent?

hzeller, 2019-12-03:
But the illegal instruction mentioned by @tgorochowik indicates that there is something wrong with that compilation.

hzeller, 2019-12-04:
There is also a `bin/unittests` in the `build/` directory; we could use that as a litmus test that things work well.

MikePopoloski, 2019-12-04:
Let me know if I can help in some way.

hzeller, 2019-12-04:
Here is the recipe how the binary is built if you want to have a look @MikePopoloski 
  https://github.com/SymbiFlow/conda-packages/tree/master/slang
it uses gcc-8 and just does a simple blaze build.

hzeller, 2019-12-04:
Current binary is here: https://anaconda.org/symbiflow/slang

And indeed, running it results in a segmentation fault. Running with gdb results in the following stacktrace; could it be some undefined initialization sequence of sorts, @MikePopoloski ?
```
gdb --args ~/conda/bin/slang-driver tests/chapter-5/5.10-structure-arrays.sv --ast-json -
GNU gdb (Debian 8.3.1-1) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ~/conda/bin/slang-driver...
(gdb) run
Starting program: ~/conda/bin/slang-driver tests/chapter-5/5.10-structure-arrays.sv --ast-json -

Program received signal SIGSEGV, Segmentation fault.
__pthread_rwlock_unlock (rwlock=0x7fffffffde80) at pthread_rwlock_unlock.c:38
38      pthread_rwlock_unlock.c: No such file or directory.
(gdb) bt
#0  __pthread_rwlock_unlock (rwlock=0x7fffffffde80) at pthread_rwlock_unlock.c:38
#1  0x0000000000551da3 in std::__shared_mutex_pthread::unlock (this=<optimized out>) at /usr/include/c++/8/shared_mutex:331
#2  std::shared_mutex::unlock (this=<optimized out>) at /usr/include/c++/8/shared_mutex:331
#3  std::unique_lock<std::shared_mutex>::unlock (this=0x7fffffffd810, this=0x7fffffffd810) at /usr/include/c++/8/bits/std_mutex.h:323
#4  std::unique_lock<std::shared_mutex>::~unique_lock (this=0x7fffffffd810, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/std_mutex.h:232
#5  slang::SourceManager::cacheBuffer (this=0x7fffffffde80, path=..., includedFrom=..., buffer=...)
    at /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/slang_1575394651870/work/source/text/SourceManager.cpp:457
#6  0x00000000005524d8 in slang::SourceManager::openCached (this=0x7fffffffde80, fullPath=..., includedFrom=...)
    at /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/slang_1575394651870/work/source/text/SourceManager.cpp:444
#7  0x0000000000552629 in slang::SourceManager::readSource (this=this@entry=0x7fffffffde80, path=...) at /usr/include/c++/8/bits/char_traits.h:287
#8  0x0000000000432ea1 in driverMain<char**> (argc=<optimized out>, argv=<optimized out>) at /usr/include/c++/8/new:169
#9  0x0000000000695b86 in generic_start_main ()
#10 0x0000000000696175 in __libc_start_main ()
#11 0x000000000041d789 in _start () at /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/slang_1575394651870/work/tools/driver/driver.cpp:431
```

mithro, 2019-12-04:
I wonder if it is because we are statically linking the binary?

mithro, 2019-12-04:
Could it be https://stackoverflow.com/questions/35116327/when-g-static-link-pthread-cause-segmentation-fault-why ?

hzeller, 2019-12-04:
it mostly looks like a not properly initialized mutex, something like some `static` storage allocated mutex without the `PTHREAD_MUTEX_INITIALIZER` initialization maybe ?

tgorochowik, 2019-12-04:
I reproduced it locally using docker and it is indeed happening only for a static build.

One of the suggested fixes in the thread linked by @mithro seems to help too, see: https://github.com/SymbiFlow/conda-packages/pull/63

closed 2019-12-04 by mithro

tgorochowik, 2019-12-04:
Just to confirm: with the slang package rebuilt and sv-tests pipeline restarted, slang is now properly tested and included in the report: https://symbiflow.github.io/sv-tests/

MikePopoloski, 2019-12-04:
Nice, thanks for tracking this down!