chipsalliance/sv-tests #487: Still work left to do after finishing make test

mithro, 2019-12-03:
In the pull request #471 I added the following test which checks there is nothing to do after running a make;
https://github.com/SymbiFlow/sv-tests/blob/55d5b7094440478da089f42ca63d683bf721c150/.github/kokoro/run.sh#L44-L69

However, `make USE_ALL_RUNNERS=1 -j$CORES --keep-going tests` completed successfully, then a `make USE_ALL_RUNNERS=1 --question tests` said there was still stuff left to run. Running the command caused a bunch of new stuff to happen!?

See the output below;
```
cgexec -g memory,cpu:sv-tests/Sv2v_zachjs ./tools/runner --runner Sv2v_zachjs --test chapter-6/6.20.2--parameter_port_list.sv --out ./out//logs/Sv2v_zachjs/chapter-6/6.20.2--parameter_port_list.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Sv2v_zachjs ./tools/runner --runner Sv2v_zachjs --test chapter-6/6.20.6--const.sv --out ./out//logs/Sv2v_zachjs/chapter-6/6.20.6--const.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Sv2v_zachjs ./tools/runner --runner Sv2v_zachjs --test chapter-6/6.20.2--parameter_dep.sv --out ./out//logs/Sv2v_zachjs/chapter-6/6.20.2--parameter_dep.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Sv2v_zachjs ./tools/runner --runner Sv2v_zachjs --test chapter-6/6.19.2--enum_sequence.sv --out ./out//logs/Sv2v_zachjs/chapter-6/6.19.2--enum_sequence.sv.log --quiet
Script done, file is /tmpfs/tmp/tmp.0LqxW6DdJg
----------------------------------------
Successful test running.
----------------------------------------
----------------------------------------
Tasks still left to run after success?
----------------------------------------
cgexec -g memory,cpu:sv-tests/Slang ./tools/runner --runner Slang --test chapter-7/arrays/packed/variable-slice-zero.sv --out ./out//logs/Slang/chapter-7/arrays/packed/variable-slice-zero.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Slang ./tools/runner --runner Slang --test chapter-7/structures/packed/default-value.sv --out ./out//logs/Slang/chapter-7/structures/packed/default-value.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Slang ./tools/runner --runner Slang --test chapter-5/5.10-structure-arrays-illegal.sv --out ./out//logs/Slang/chapter-5/5.10-structure-arrays-illegal.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Slang ./tools/runner --runner Slang --test chapter-22/22.7--timescale-basic-4.sv --out ./out//logs/Slang/chapter-22/22.7--timescale-basic-4.sv.log --quiet
cgexec -g memory,cpu:sv-tests/Slang ./tools/runner --runner Slang --test chapter-22/22.7--timescale-basic-3.sv --out ./out//logs/Slang/chapter-22/22.7--timescale-basic-3.sv.log --quiet
```

hzeller, 2019-12-03:
Maybe for whatever reasons, log-files were not generated initially ? Could there be a sequencing problem of log directories generated vs. storing logs in there ?
I noticed that after recent changes (the cgroup change ?) a simple `make` does not work anymore, as it fails to write to files. One has to explicitly invoke `make tests` in order to get the tests running, so this might be an indicator.

When make is run a third time, does it still make things again ?

hzeller, 2019-12-03:
Hypothesis: It could also be that the logs are not generated because they are skipped tests (e.g. simulation tests for tools that don't support it).

hzeller, 2019-12-19:
Fixed by #521

closed 2019-12-19 by hzeller