chipsalliance/sv-tests #548: Add Verilator in multiple modes

mithro, 2020-01-10:
It would be good to add Verilator in;
 - [ ] Parse only mode
 - [ ] simulation mode
 - [ ] pedantic / strict mode
 - [ ] other?

mithro, 2020-01-10:
#506 is related

mithro, 2020-01-10:
FYI - @wsnyder

mithro, 2020-01-10:
FYI - @taylor-bsg

wsnyder, 2020-01-10:
Verilator has

* Normal simulation, "-cc --Wpedantic" etc
* Parsing "--xml --bbox-unsup --Wpedantic"
* Lint only "--lint-only --bbox-unsup --Wpedantic"

For all of these, given this test suite's assumptions, I would retain the pedantic flag.

Note that --xml mode still does elaboration so will report errors on bad code in many cases where e.g. sv_parser will not.

I'd say it's up to the group how many of these you want tested. My personal interest is the simulation mode.  After that parsing has the most language support, and lint is in the middle.  I don't think the test cases have a flag to indicate what should pass for a linter, which makes that a bit more complicated. So if adding something would probably suggest to add parsing/xml first.

A minor complication is the test report is getting wide so it's hard to see the ones at the end of the line at the same time as titles at the left; perhaps some javascript scroll hackery is needed, or as a stopgap maybe put the titles at both left and right edges.

taylor-bsg, 2020-01-10:
So is lint closest to parse only, then?

M

wsnyder, 2020-01-10:
XML is closest to parse only.  Lint is half way between XML and simulation.

wsnyder, 2023-08-31:
The difference between the various modes is much smaller than it used to be, as verilator's now simulating 90% of the test cases. I suggest we can close this out as not needed.  (I can't close it myself.)