chipsalliance/sv-tests #549: Some basejump_stl files are abstract base classes and should be marked parse only

mithro, 2020-01-10:
Apparently some of the files are like "abstract base classes" so should be marked as parse only. 

Verilator is failing on these because it parses it and then sees the abstract nature and correctly fails.

Apparently there is a bsg_micro repo which instantiates these ABC in a bunch of different ways. Currently it is converted to pure Verilog with design compiler and bsg-sv2v - we need to get them to publish the unconverted version.

mithro, 2020-01-10:
FYI - @taylor-bsg

taylor-bsg, 2020-01-10:
+ @wsnyder Yes, generally if you grep for "inv", with quotes, those are abstract modules that require parameter instantiation.  Those modules should be parsable by verilator, but should fail in elaboration. It would be nice to run Verilator in parse-only mode on these, if such a thing exists.

mithro, 2020-01-10:
@taylor-bsg - I can't find this magical bsg_micro repository?

taylor-bsg, 2020-01-10:
The repo is https://github.com/bsg-idea/bsg_micro_designs, but this repo contains code that have already been elaborated by https://github.com/bespoke-silicon-group/bsg_sv2v, it is not SystemVerilog. With some work, we could possibly do bsg_micro_designs_sv, which does not run it through bsg_sv2v.

mithro, 2020-01-10:
@taylor-bsg - Why do you have @bsg-idea and @bespoke-silicon-group!?

mithro, 2020-01-10:
@taylor-bsg - You must have a source repo for the stuff before it goes through sv2v right?

taylor-bsg, 2020-01-10:
bsg-idea versus bespoke-silicon-group -- basically because github does not
have hierarchy, we are trying to avoid having organizations with 100 repos
jumped together. We also have an org for external repos that we have
forked, and an org for black-parrot. If you have a better way to do it I
would love to know!