chipsalliance/sv-tests #560: The UVM test is not a proper test

alainmarcel, 2020-01-11:
The UVM test is including the uvm_pkg.sv which is not the proper usage. It basically forces Surelog to reparse the package, hence the timeout.
The UVM package come precompiled with Surelog.
Proper use of the uvm_pkg is actually to be parsed as one of the first file in the command line,
then subsequent files that need to use the elements of the package use the import statement and `include "uvm_macros.svh"
See https://github.com/alainmarcel/Surelog/blob/master/tests/SimpleClass1/SimpleClass1.sl for proper test.

hzeller, 2020-07-12:
Looking at all the UVM tests, this looks like it is fixed in the meantime.
https://symbiflow.github.io/sv-tests/#Surelog|uvm-agents|uvm_agent_active

Ready to close @alainmarcel ?

tgorochowik, 2021-08-05:
Let's close it, there have been multiple UVM tests improvements since this issue was created

closed 2021-08-05 by tgorochowik