chipsalliance/sv-tests #577: Undefined macro usage

dalance, 2020-01-22:
I'm wondering whether `third_party/tests/ivtest/ivltests/undef.v` should be parsed or not.
The simplified code is below:

```
`define a 1
`undef a
`a
```

My sv-parser and almost tools can't parse it because `a` is not defined.
But the comment of the test says below:

```
/*
 * 1364-2001 19.3.2 "An undefined text macro has no value, just as if it had
 * never been defined."
 */
```

The author of the test seems to intend that `a` has no value and `a` should be replaced to empty string without error.

What do you think?

Thomasb81, 2020-01-25:
I think 

> /*
 > * 1364-2001 19.3.2 "An undefined text macro has no value, just as if it had
>  * never been defined."
>  */

refer to this use-case:

```
`undef a
`ifdef a
wire my_net;
`else
reg my_net;
`endif
```

What the point to do a macro substitution of something not defined ?
The test is may be is expected to fail in order to test tool error report mechanic...

dalance, 2020-01-27:
The simplified code may be confusing...
The original test is below:

```verilog
/*
 * 1364-2001 19.3.2 "An undefined text macro has no value, just as if it had
 * never been defined."
 */
`define a 1

`ifdef a
`define b 1
`else
`define b 0
`endif

`undef a

`ifdef a
`define c 1
`else
`define c 0
`endif

module test;
initial begin
    if(`a+1 !== 1) begin $display("FAIL"); $finish; end
    if(`b+1 === 1) begin $display("FAIL"); $finish; end
    if(`c+1 !== 1) begin $display("FAIL"); $finish; end

    $display("PASSED");
end
endmodule
```

If the usage of undefined macro (`` `a+1 !== 1``) is expected to fail, the remaining code is meaningless.
Actually Icarus and Verible pass the test, so the authors of the tools seem to think the usage of undefined macro is valid.

Thomasb81, 2020-01-27:
Well my copy of Ieee1800-2017 is not more explicit... 

But Icarus and Verible behavior bother me, because it would mean consecutive macro directive  :
```
`define a something
`undef a
``` 
and

```
`define a something
`define a
``` 
are equivalent.

While
```
`define a something
`undef a
` ifdef a
/* something */
`else
/* something else */
`endif
``` 
and 
```
`define a something
`define a
` ifdef a
/* something */
`else
/* something else */
`endif
``` 

are not. 

It would be interesting to compare the behavior with commercial tool and also with a C pre-processor.

piotr-binkowski, 2020-02-03:
I agree that IEEE1800-2017 is not clear on what should happen when accessing an undefined macro and maybe the best solution would be to blacklist `ivtest/ivltests/undef.v` test so that we won't penalize any tool for something that is not clearly defined in SystemVerilog standard.

dalance, 2020-02-03:
Thanks.
I'll add `undef.v` to blacklist.

closed 2020-02-03 by piotr-binkowski