chipsalliance/sv-tests #580: BlackParrot Verilog vs SystemVerilog files

alainmarcel, 2020-01-23:
I logged the following issue on BlackParrot, but I want to make sure sv-tests does the right thing when creating the project file.

On SV-Tests, I have the following error (In Surelog) on Blackparrot, "logic" is a SystemVerilog construct, you either need to make your file .sv or put a -sv prior to the file or -sverilog on the overall command line to treat all the files as SysteVerilog files. Right now, it is a .v file which should follow the Verilog syntax (which I flag).
SYNTX:PA0207] third_party/cores/blackparrot/bp_be/src/v/bp_be_calculator/bp_be_int_alu.v:58 Syntax error: no viable alternative at input 'logic signed',logic signed [reg_data_width_lp-1:0] src1_sgn , src2_sgn; ^-- ./slpp_all/work/third_party/cores/blackparrot/bp_be/src/v/bp_be_calculator/bp_be_int_alu.v:59 col:6.[WARNI:PA0205] third_party/cores/blackparrot/external/basejump_stl/bsg_noc/bsg_noc_pkg.v:4 No timescale set for "bsg_noc_pkg".

taylor-bsg, 2020-01-23:
To clarify Alain's issue; it appears that sv-tests is not using his required -sverilog flag to indicate that BlackParrot is is systemverilog files.

alainmarcel, 2020-01-23:
.v files are Verilog files unless specified otherwise by the mean of command line options, -sv filename or -sverilog are the most common ways to overwrite the default parsing in Verilog format. .sv files are SystemVerilog files. BlackParrot does not follow the convention, somewhere either in BlackParrot Makefile or in the sv-test this needs to be fixed

alainmarcel, 2020-01-29:
sv-test has to use the file lists from the respective projects and make sure proper file type identification is done.