chipsalliance/sv-tests #581: Add tool using ANTLR syntax

jwagen, 2020-01-23:
There exist an ANTLR description of the SystemVerilog syntax.
 [https://github.com/antlr/grammars-v4/tree/master/verilog/system_verilog](https://github.com/antlr/grammars-v4/tree/master/verilog/system_verilog)

From the readme it appears to be a project Microsoft research did a couple of years ago. I am not that well versed in ANTLR so I dont know how much would be required to turn this into an executable tool and how useful it would be to run the tests against it. It looks like it does not have the grammar for `class` and many other verification constructs.

The antlr4 documentation describe how to create a parser from the grammar file for multiple languages. For python it seams rather simple.
[https://github.com/antlr/antlr4/blob/master/doc/python-target.md](https://github.com/antlr/antlr4/blob/master/doc/python-target.md)

There is also an extension for Visual Studio code that seems to have adopted antlr files for syntax checking. I dont know if they have the same origin or if they are written from scratch. The syntax files in the plugin supports constructs like class which is not in the grammar in the ANTLR repo. There is quite a bit of activity in the repo, and some of the effort looks like it is toward creating a language server.
[https://github.com/eirikpre/VSCode-SystemVerilog](https://github.com/eirikpre/VSCode-SystemVerilog)
[https://github.com/eirikpre/VSCode-SystemVerilog/tree/master/src/compiling/ANTLR/grammar](https://github.com/eirikpre/VSCode-SystemVerilog/tree/master/src/compiling/ANTLR/grammar)

Again I don't know how useful this is as neither of these seems to exist as standalone tools.

alainmarcel, 2020-01-23:
Surelog is ANTLR based and is running in SVtest

alainmarcel, 2020-01-24:
https://github.com/alainmarcel/Surelog

jwagen, 2020-01-28:
Oh. I must have missed the addition of Surelog to the tests. It looks like the project is in a way better shape than the one I mentioned.
Great work by the way, creating a parser with such high coverage in the tests.

I guess this issue can be closed then.

mithro, 2020-01-28:
@jwagen - I'm sure @alainmarcel would love people to try out SureLog (and potentially even contribute?).

closed 2020-01-28 by mithro

alainmarcel, 2020-01-29:
Sure thing!