chipsalliance/sv-tests #633: slang version in published report is nearly two months old

MikePopoloski, 2020-02-08:
I'm not sure why because both this repository and the conda package repository were updated, but it would be nice if the report used an up to date binary.

kgugala, 2020-02-09:
@MikePopoloski indeed conda builds an pushes the latest slang packages, but they are not used by the `sv-tests`. This is because you have introduced `nightly` tagging in the slang's CI. This tag is used to name the conda package effectively changing the package name from `slang` to `slang-nightly`. Conda package builder expects the tags in format `v.X` where `X` is the version number. Since `sv-tests` is using `slang` conda package it always gets the latest of this name, which happen to be a more than 2 month old one. 

I think we can switch `sv-tests` to use `slang-nightly` package. This should solve the issue.

MikePopoloski, 2020-02-09:
Ah, that makes sense, thanks for the explanation!

piotr-binkowski, 2020-02-10:
I've checked this and this appears to be a bit more complicated, `nightly` build still ends up in the `slang` conda package as can be seen [here](https://anaconda.org/SymbiFlow/slang/files) but conda determines the latest package on file name only which causes it to use the older package. I will need to fix this in the [conda-packages](https://github.com/SymbiFlow/conda-packages) repo.

MikePopoloski, 2020-02-29:
Seems like this was completed. Thanks guys!

closed 2020-02-29 by MikePopoloski