chipsalliance/sv-tests #746: How to add a proprietary/commercial tool?

krupan, 2020-04-12:
I understand that we won't be able to publicly test proprietary/commercial tools, but I'd love to run this suite against a commercial tool or two in the privacy of my own home/business.  It is not possible to make a commercial tool available on Anaconda...is it?

tgorochowik, 2020-04-21:
Hello krupan! Thank you for your interest in this project!

We do not support testing commercial tools at all.

That being said, if you were to test your own tool for which you do not have a conda package, you can simply skip that step - the conda packages are only required for CI (so we don't have to build them here). If your own tool is available in your PATH or you use absolute path in your runner wrappers, it should work just fine.

dalance, 2020-04-21:
For example, I use the runner wrapper for Synopsys VCS like below.
If the following code is added as `tools/runners/vcs.py` it will work fine.

```
from BaseRunner import BaseRunner

class vcs(BaseRunner):
    def __init__(self):
        super().__init__("vcs", "vlogan")

        self.url = ""

    def prepare_run_cb(self, tmp_dir, params):
        self.cmd = [self.executable]
        self.cmd.append('-full64')
        self.cmd.append('-sverilog')

        for incdir in params['incdirs']:
            self.cmd.append('+incdir+' + incdir)

        self.cmd += params['files']
```

closed 2021-04-21 by tgorochowik