chipsalliance/sv-tests #749: Add license header to top of all code files

mithro, 2020-04-13:
See https://opensource.google/docs/releasing/licenses/

We can either use;
```
Copyright (C) 2020  The SymbiFlow Authors.

Use of this source code is governed by a ISC-style
license that can be found in the LICENSE file or at
https://opensource.org/licenses/ISC
```

Or put the full license at the top;
```
Copyright (C) 2020  The SymbiFlow Authors.

Permission to use, copy, modify, and/or distribute this software for any
purpose with or without fee is hereby granted, provided that the above
copyright notice and this permission notice appear in all copies.

THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES
WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF
MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR
ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES
WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN
ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF
OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.
```

wsnyder, 2020-04-16:
Suggest you consider including SPDX-License-Identifier tags: https://spdx.org/ids-how

mithro, 2020-04-16:
@wsnyder - Yeap! -> Full list of things is at https://github.com/SymbiFlow/ideas/issues/47

PR #762 (2020-12-23): Add license header to top of all code files
Fixes #749
commit 4ef2944c (2020-12-23): Add license header to top of all code files

Fixes #749
closed 2020-12-23 by tgorochowik