chipsalliance/sv-tests #763: sv-report fails if test uses nonexistent source files

tgorochowik, 2020-04-21:
Also see: #748

Formatting the source files:
https://github.com/SymbiFlow/sv-tests/blob/d60466dd1b2b778bfd67e425b4082378f227ba21/tools/sv-report#L106-L125

does not have any exception handling for when the input file doesn't exist.
I think it is okay to fail if we cannot write the formatted file (however with a nicer error probably), but if we cannot read the source file (e.g. it doesn't exist) we should create the output file with a message like ("source file doesn't exist") and continue.

Another option would be to fail in both cases, but with a meaningful message.

tjurtsch, 2020-06-30:
I'll do this, the way you described.
WIP PR is here https://github.com/SymbiFlow/sv-tests/pull/902

tjurtsch, 2020-07-01:
Since the PR was merged, this issue can be closed.

closed 2020-07-01 by tgorochowik