chipsalliance/sv-tests #770: Verilator tests broken recently

wsnyder, 2020-04-21:
The report (https://symbiflow.github.io/sv-tests/) hadn't been updated for a while AFAIK, but I noticed now has an update.

I was all excited to see more green boxes for Verilator given recent things like classes, but alas, there's a ton of tests that were ok around a week ago and are now broken by something.   When I run it manually on master it seems to pass.

Can someone please look into what's going on?  Thanks.

E.g. chapter-6/6.13--void.sv.log under https://symbiflow.github.io/sv-tests/

```
sh scr.sh --cc -Wno-fatal -Wno-UNOPTFLAT -Wno-BLKANDNBLK -Wpedantic -Wno-context -I/tmpfs/src/github/sv-tests/tests/chapter-6 --Mdir vbuild --prefix Vtop --exe -o vmain vmain.cpp tests/chapter-6/6.13--void.sv
scr.sh: 3: scr.sh: ./vbuild/vmain: not found

```

P.S. Is there a way to see older versions of the web reports?

mithro, 2020-04-21:
@wsnyder - You bring up a good point, we should probably be publishing a history of the reports somewhere.

mithro, 2020-04-21:
@wsnyder I think they are stored in https://github.com/SymbiFlow/sv-tests/commits/gh-pages branch but not published anywhere...

tgorochowik, 2020-04-22:
> E.g. chapter-6/6.13--void.sv.log under https://symbiflow.github.io/sv-tests/
>
> ```
> sh scr.sh --cc -Wno-fatal -Wno-UNOPTFLAT -Wno-BLKANDNBLK -Wpedantic -Wno-context [...]

This test passes for me locally - so I've created a PR to make verilator logs more verbose in the report. This should make it easier to see why it fails.

wsnyder, 2020-04-24:
The problem seems to be "ccache" is installed on some machines (namely the machine building verilator), and not all (which is why some tests randomly fail).  Defaulting to using ccache when installed was a recent Verilator change, so explains why it's breaking now.  Can you install ccache in your containers or otherwise?

If not you can suppress this with

    export OBJCACHE=

or equivalent when running Verilator, but this will waste CPU time.

commit 03726672 (2020-05-14): Install ccache to fix verilator

Fixes #770
closed 2020-05-14 by tgorochowik