chipsalliance/sv-tests #853: New uvm-agent tests are not legal; need UVM first

wsnyder, 2020-06-05:
The recently added uvm-agent tests use this sort of command line:

    verilator --cc ...  tests/testbenches/uvm_agent_active.sv third_party/tests/uvm/src/uvm_pkg.sv

This is not legal, IEEE requires all classes be declared before they are used.  This happened to exposed a verilator error handling bug, this is fixed in Verilator master, and Verilator now properly will give an error message.

Anyhow the uvm_pkg must be first on the command line, that is

    verilator --cc ...  third_party/tests/uvm/src/uvm_pkg.sv tests/testbenches/uvm_agent_active.sv

However I would suggest the nicer fix would be to edit each of the uvm agent sources to do a

     `include "uvm_pkg.sv"

at the top, as this makes it much easier to run the tests outside the sv harnesses, and makes it far more intuitive looking at the sources what is going on.

Thanks.

kgugala, 2020-06-06:
Hi @wsnyder thanks for catching this. We'll bump the Verilator to use the version with fixes and fix the pkg compliation order.

hzeller, 2020-07-12:
Looking at the log output, the pkg is now the first in the command-line
  https://symbiflow.github.io/sv-tests/#Verilator|uvm-agents|uvm_agent_active
Does this look good now @wsnyder ?

W.r.t using include directly: we try to keep the features tested as localized as possible so that we don't mix in testing if a tool supports include properly (though in this particular case it would not really make a big difference as uvm_pkg.sv in itself is mostly just including other stuff...).

In the above log, there are errors reported, but these look like genuine reports not related to the package order ?

wsnyder, 2020-07-12:
I think the logfile is reasonable, the error that indicates that a package is not declared is no longer present, so seems ok.

hzeller, 2020-07-13:
Alright, closing the issue.

closed 2020-07-13 by hzeller