chipsalliance/sv-tests #862: New class tests are not legal IEEE

wsnyder, 2020-06-06:
Thanks for adding more SV tests, these will be helpful as tools support more of UVM.

Some cleanup is needed as many of the tests are not IEEE legal.  Taking for example class_member_test_2:

    class myclass;
     extern task subtask(arg_type arg);
    endclass

This is not legal as arg_type is not declared.  This fails on all of the closed-source simulators.

Could these be cleaned up please, and ideally have a process to check tests against some simulator (parsing is insufficient) before they hit master?  

Thanks

tgorochowik, 2020-06-08:
Thanks for the report!

The tests you mention are not new - these are the tests contained in `tests/generic/` these were generated some time ago from verible before it was open sourced. Therefore they are meant to be uses to test parsers only.

I guess we should explicitly mark those tests as such.

tgorochowik, 2020-06-08:
I guess we can also consider removing individual tests from that directory - we have done that in the past for some imported tests - most of the features should already be covered by our tests anyway.

wsnyder, 2020-06-08:
Thanks for looking at this.

As long as there's similar coverage, removing seems reasonable, but there still has to be tests left in that call UVM. Those tests either need a include of UVM and runner to have the lib feature removed (preffered), or have runner include libs in the proper order. Either way note runner needs a small fix also.

PR #879 (2020-06-18): Tests cleanup
Removed/modified tests which have illegal IEEE syntax or are already
covered in other tests.

Fixes #862
closed 2020-06-18 by symbiflow-robot