chipsalliance/sv-tests #863: third_party UVM submodule?

wsnyder, 2020-06-08:
sv-tests' third_party/tests/uvm is not a submodule.

Can we make it a submodule so that I and others can clone it for standalone Verilator testing etc? 

Assuming you agree this makes sense, where should it be?
  a. Chips Alliance? Upside is it's an obvious base for other CHIPS projects to include, downside is it's not really something chips "owns"
  b. Verilator?  Currently UVM is usually distributed by the simulator vendors, so this seems appropriate, and also we will eventually want some `ifdef VERILATOR lines in there to improve some performance issues.
  c. Other?

Regardless of a/b/c I would ideally like it in a place where I have commit permission so can maintain a branch holding the subset that Verilator can currently parse (all but ~63 lines).

tgorochowik, 2020-06-08:
At the moment `third_party/tests/uvm` is just the code extracted from the archive Accellera provides (as mentioned in this [commit message](https://github.com/SymbiFlow/sv-tests/commit/b70882180c1df9f32e22c18bc677403ed272612b)).

Or do you want all the individual uvm tests to be moved out of the tree? For now they are in the `tests/` directory as they aren't really "third party", see an example [here](https://github.com/SymbiFlow/sv-tests/blob/6efc94ec4ba45c24bf4d97619310a4a1f097b383/tests/chapter-16/16.10--property-local-var-uvm-fail.sv).

I guess this could be moved to a separate repository if it would be in any way useful. One issue we would have to solve is moving the test generators out of the tree too as some of the uvm tests are generated from [templates](https://github.com/SymbiFlow/sv-tests/blob/686ba0807a28a3e614a09120b1ee55c9bcb9072d/conf/generators/templates/uvm-classes_0.sv).

wsnyder, 2020-06-08:
I'm interested in having third_party_tests/uvm as a repository (like easyUVM), so it can be a submodule of sv-tests, of verilator_ext_tests, and of people using it in production for their own code (e.g. eventually SweRV will have tests and need the submodule),

I think the main question is where should it live?   E.g. if I add a verilator/uvm would you be willing to point to it?

wsnyder, 2020-06-09:
#865 if acceptable is a pull to do what I'm suggesting.

tgorochowik, 2020-06-10:
The PR is merged, closing, thanks!

closed 2020-06-10 by tgorochowik