chipsalliance/sv-tests #903: Import XilinxUnisimLibrary as a test suite?

mithro, 2020-06-30:
Does it make sense to import the https://github.com/SymbiFlow/XilinxUnisimLibrary library as a third_party test suite?

mithro, 2020-06-30:
@hzeller @tgorochowik -- Thoughts? Is the UnisimLibrary pure verilog?

hzeller, 2020-07-10:
@alainmarcel has it in [the Surelog testuite](https://github.com/alainmarcel/Surelog/tree/master/third_party/tests/XilinxUnisimLibrary) in the meantime ( https://github.com/alainmarcel/Surelog/issues/568 ).

Would be worthwhile to include in sv-tests

hzeller, 2020-07-13:
Assigning to Karol for now, please delegate to whoever might be free. Having it in sv-tests will help us see how the tools deal with it (.. and/or how changes in the Unisims files change that) 

Here is how they are run in Surelog (they are there in a batchfile, but it is easy to see each individual line is one invocation)
  https://github.com/alainmarcel/Surelog/blob/master/third_party/tests/XilinxUnisimLibrary/batch.txt

Since we need unique names per test, maybe we just compose it from the filename with some prefix in the generate script. So file `RX_BITSLICE.v` would become test `unisims-RX_BITSLICE` maybe.

mithro, 2020-07-22:
@alaindargelas - The upstream sim models has been updated with the missing `verilog/src/glbl.v` file which should make it possible to get further along in the parsing.

alaindargelas, 2020-07-22:
Updated, also fixed my grammar for the 2 syntax errors => 0 Syntax errors. All files are parsed 100%.