chipsalliance/sv-tests #915: Classification of test to should fail. Wrong application of json file ?

hzeller, 2020-07-10:
The utd-tests are currently mostly classified as failing.
[image]

.. with a suspicious should-fail message *imported from yosys repository contains intentional syntax error*

This string shows up only in `conf/generators/meta-path/yosys-errors.json`. Is the wrong json file applied while generating the utd tests ?

Assigning to @tjurtsch as he was last editing the yosys json file (though the problem is probably not in the file but in the generator).

hzeller, 2020-07-10:
Ok, assigning to @tgorochowik for now as @tjurtsch does not show up yet in the bug-assignment list (do we need to add him @mithro ?)

tgorochowik, 2020-07-10:
One extra thing I've noticed is that we have an empty `should_fail_because:` for tests that are expected to pass. This doesn't seem to break anything but looks confusing, so we should probably get rid of that too.

hzeller, 2020-07-10:
This example above now moved to basejump:
[image]

Could this be because here all json files are processed in *.json order, which is essentially random depending on how the inodes are arraged ? Should we use *.json in the first place or maybe more specific ?

piotr-binkowski, 2020-07-10:
I just noticed that this may be caused by the fact that we only clear `should_fail_because` once here https://github.com/SymbiFlow/sv-tests/blob/master/generators/path_generator#L28 which is outside of the json processing loop so next json file processed will use the message from previous one even if it shouldn't fail @tgorochowik @tjurtsch

hzeller, 2020-07-10:
That looks like it explains what we see @piotr-binkowski

PR #919 (2020-07-10): Clear should_fail_because for each template run
As suggested by @piotr-binkowski

Fixes #915
commit 8181b9fb (2020-07-10): Clear should_fail_because for each template run (#919)

As suggested by @piotr-binkowski

Fixes #915
closed 2020-07-10 by hzeller