chipsalliance/sv-tests #932: Report page: switch on Jinja whitespace removal ?

hzeller, 2020-07-12:
The report page is nicely formatted, but also contains a lot of whitespace that unnecessarily has to be transmitted (the whole page is in the order of 10MiB).

I think Jinja has an option to trim unnecessary whitespace for more compact output.

tgorochowik, 2020-07-24:
I ran some tests (note that I used only 2 runners to speed up the process, but I am giving the results in percents in hope it reflects running all runners):
* forcing jinja to hide whitespace where it can (substituting `{% %}` with `{%- -%}` etc) saves only about 4%
* jinja is not a html formatter so it still mostly preserves the whitespace from the templates, it can only control whitespace around it's dynamic statemets
* Using obfuscation I was able to save ~39% - the added benefit of this approach is it can also make js and css smaller. I know obfuscation is not what we really want here, but smaller size is a nice benefit of using obfuscation so I think it's worth it

I will create a PR to see what we get for the whole report.

tgorochowik, 2020-07-24:
When running some tests I noticed that some files for the uhmd-yosys take ~60MB(!) (for example: https://github.com/SymbiFlow/sv-tests/blob/f25bce50e27b2365574f0df735f297ddfa204403/logs/UhdmYosys/generated/black-parrot/bp_processor.sv.log.html )

This is mostly due to dumping a horribly long uhdm tree. I asked @kbieganski to add a switch to the uhdm-yosys integration to be able to disable dumping the tree (this is something we already implement for uhdm-verilator)

kbieganski, 2020-07-27:
The latest version of Yosys (antmicro/yosys@490d671) does not print the tree by default anymore, so the output size should be reduced significantly.

tgorochowik, 2020-07-27:
@kbieganski great, thanks!

tgorochowik, 2022-01-04:
The main page has been optimized #1941, it's now about 300KiB. Let's close this.

closed 2022-01-04 by tgorochowik