chipsalliance/sv-tests #955: slang build is a couple months old

MikePopoloski, 2020-07-23:
I raised this as an issue in the conda project, but I'm not sure if anyone looks at that -- it seems like the slang build has been failing there due to some config / build system issues there and so hasn't been updated for the tests in a while. I'd appreciate if someone could help look at it, it doesn't appear to be something on my end at first glance. Happy to help too if you can point me in the right direction.

tgorochowik, 2020-07-23:
@MikePopoloski I am sorry we missed the original issue.

We generally had to patch slang in conda because there were some incompatibilities with the libs provided by conda (I don't remember the details, we'll have to look it up).

The solution was to patch slang to build it statically. The patch currently fails to apply after some changes in upstream.

Hopefully this will help:
https://github.com/SymbiFlow/conda-packages/pull/125

If it does - would you consider adding a separate build target to slang, to build it statically? If that was added we could use that target for the conda package to make sure it does not occur again when your CMake files change

MikePopoloski, 2020-07-23:
I see, yeah, I'll look into adding another target so that you don't need to maintain the patch. Thanks for your help!

closed 2020-07-23 by mithro