chipsalliance/sv-tests #969: always check has infinite loop

caryr, 2020-07-26:
The following code from the test suite has an infinite loop since the always does not have any delay. I would suggest adding a sensitivity to the always (e.g. @(a)) to avoid this infinite loop. Icarus finds and reports this infinite loop and fails instead of allowing the user to run code that will lock up.
```
/*
:name: always
:description: always check
:tags: 9.2.2.1
*/
module always_tb ();
	wire a = 0;
	reg b = 0;
	always
		b = ~a;
endmodule
```
Also keep in mind Icarus is a two stage tool so parsing and elaboration are done in the first stage. It look like a bunch of the other failures reported for Icarus are because it is not just parsing the files, but is also trying to elaborate them. If you are interested I can make suggestions that should allow Icarus to process more of the checks correctly and still test things as expected.

I also have some ideas how to make the ivtest suite a bit more compatible. Many of the skipped tests are because we use a language generation flag to parse the files with the appropriate keywords (language generation). If we also add a compiler directive this should fix the issue when SystemVerilog is assumed.

caryr, 2020-11-15:
Any update on this?

tgorochowik, 2020-11-16:
Thanks for the report! I am sorry we missed this.

I think we shouldn't be adding a sensitivity list to this test as it was supposed to test the `always` blocks without it.
You are right that this creates a deadlock though. I think that we should add a delay to the block and extend the tags list to indicate that it expects delays too.

PR #1144 (2020-11-23): Remove deadlock from `always` test
Fixes #969
commit 6c2c6472 (2020-11-23): Remove deadlock from `always` test

Fixes #969
closed 2020-11-23 by tgorochowik