chipsalliance/sv-tests #1143: Error in test code

caryr, 2020-11-15:
The file tests/chapter-5/5.8-time-literals.sv has an error that Icarus is correctly reporting. On line 23 there is a ';' after the end and that is not supported by the SV standard. If this is supported please provide a reference to where this is described in the standard.

MikePopoloski, 2020-11-15:
It is allowed, see the grammar in Annex A:
```
A.1.4 Module items
module_common_item ::= module_or_generate_item_declaration | ...
module_or_generate_item_declaration ::= package_or_generate_item_declaration | ...

A.1.11 Package items
package_or_generate_item_declaration ::=
  net_declaration
| data_declaration
| task_declaration
| function_declaration
| checker_declaration
| dpi_import_export
| extern_constraint_declaration
| class_declaration
| class_constructor_declaration
| local_parameter_declaration ;
| parameter_declaration ;
| covergroup_declaration
| assertion_item_declaration
| ;
```

caryr, 2020-11-15:
Thank you for the reference! I will need to look at adding this to Icarus.

closed 2020-11-15 by caryr

caryr, 2020-11-21:
I have updated Icarus to support this so the test should no longer fail.