chipsalliance/sv-tests #1146: Avoid UVM on chapter 18 tests?

wsnyder, 2020-11-20:
There's many tests in chapter 18 that require UVM.  It seems misleading to users, and discouraging to us tool authors, to have non-green row for e.g. $urandom (18.13.1) for a simulator that supports $urandom but not UVM because 18.13.1--urandom_3.sv fails due to no UVM support.

There's a number of options:

0. Do nothing, it is what it is :)

1. Delete UVM test when there's a corresponding non-UVM test of similar capability.

2. Have the test tag line not list sections other than UVM if UVM is used. e.g. change "tags: 18.13.1 uvm" to "uvm".

3. Other?

If there's no other opinion 2, remove section tag, is my suggestion.  If agreeable I can provide a pull request.

Relevant tests (Some don't have corrsponding non-UVM tests):

```
  18.5--constraint-blocks_1.sv:4::tags: 18.5 uvm
  18.5.1--explicit-external-constraint_2.sv:4::tags: 18.5.1 uvm
  18.5.1--implicit-external-constraint_2.sv:4::tags: 18.5.1 uvm
  18.5.11--static-constraint-blocks_1.sv:4::tags: 18.5.11 uvm
  18.5.12--functions-in-constraint_1.sv:4::tags: 18.5.12 uvm
  18.5.13--constraint-guards_1.sv:4::tags: 18.5.13 uvm
  18.5.14--soft-constraints_1.sv:4::tags: 18.5.14 uvm
  18.5.14.1--soft-constraint-priorities_1.sv:4::tags: 18.5.14.1 uvm
  18.5.14.1--soft-constraint-priorities_3.sv:4::tags: 18.5.14.1 uvm
  18.5.14.1--soft-constraint-priorities_4.sv:4::tags: 18.5.14.1 uvm
  18.5.14.2--discarding-soft-constraints_1.sv:4::tags: 18.5.14.2 uvm
  18.5.14.2--discarding-soft-constraints_3.sv:4::tags: 18.5.14.2 uvm
  18.5.14.2--discarding-soft-constraints_5.sv:4::tags: 18.5.14.2 uvm
  18.5.2--constraint-inheritance_1.sv:4::tags: 18.5.2 uvm
  18.5.2--pure-constraint_1.sv:4::tags: 18.5.2 uvm
  18.5.3--set-membership_1.sv:4::tags: 18.5.3 uvm
  18.5.4--distribution_1.sv:4::tags: 18.5.4 uvm
  18.5.5--uniqueness-constraints_1.sv:4::tags: 18.5.5 uvm
  18.5.6--implication_1.sv:4::tags: 18.5.6 uvm
  18.5.7--if-else-constraints_4.sv:4::tags: 18.5.7 uvm
  18.5.8.1--foreach-iterative-constraints_1.sv:4::tags: 18.5.8.1 uvm
  18.5.8.2--array-reduction-iterative-constraints_1.sv:4::tags: 18.5.8.2 uvm
  18.5.9--global-constraints_1.sv:4::tags: 18.5.9 uvm
  18.6.1--randomize-method_0.sv:4::tags: 18.6.1 uvm
  18.6.2--post-randomize_method_1.sv:4::tags: 18.6.2 uvm
  18.6.2--pre-randomize-method_1.sv:4::tags: 18.6.2 uvm
  18.6.3--behavior-of-randomization-methods_1.sv:4::tags: 18.6.3 uvm
  18.6.3--behavior-of-randomization-methods_2.sv:4::tags: 18.6.3 uvm
  18.6.3--behavior-of-randomization-methods_3.sv:4::tags: 18.6.3 uvm
  18.6.3--behavior-of-randomization-methods_5.sv:5::tags: 18.6.3 uvm
  18.7--in-line-constraints--randomize_0.sv:4::tags: 18.7 uvm
  18.7--in-line-constraints--randomize_2.sv:4::tags: 18.7 uvm
  18.7--in-line-constraints--randomize_4.sv:4::tags: 18.7 uvm
  18.7--in-line-constraints--randomize_6.sv:4::tags: 18.7 uvm
  18.7.1--local-scope-resolution_1.sv:4::tags: 18.7.1 uvm
  18.8--disabling-random-variables-with-rand_mode_0.sv:4::tags: 18.8 uvm
  18.8--disabling-random-variables-with-rand_mode_1.sv:4::tags: 18.8 uvm
  18.8--disabling-random-variables-with-rand_mode_2.sv:4::tags: 18.8 uvm
  18.8--disabling-random-variables-with-rand_mode_3.sv:4::tags: 18.8 uvm
  18.8--disabling-random-variables-with-rand_mode_5.sv:5::tags: 18.8 uvm
  18.9--controlling-constraints-with-constraint_mode_0.sv:4::tags: 18.9 uvm
  18.9--controlling-constraints-with-constraint_mode_2.sv:5::tags: 18.9 uvm
  18.10--dynamic-constraint-modification_0.sv:4::tags: 18.10 uvm
  18.11--in-line-random-variable-control_0.sv:4::tags: 18.11 uvm
  18.11--in-line-random-variable-control_1.sv:4::tags: 18.11 uvm
  18.11.1--in-line-constraint-checker_0.sv:4::tags: 18.11.1 uvm
  18.11.1--in-line-constraint-checker_1.sv:4::tags: 18.11.1 uvm
  18.12--randomization-of-scope-variables_1.sv:4::tags: 18.12 uvm
  18.12.1--adding-constraints-to-scope-variables_1.sv:4::tags: 18.12.1 uvm
  18.13.1--urandom_1.sv:4::tags: 18.13.1 uvm
  18.13.1--urandom_3.sv:4::tags: 18.13.1 uvm
  18.13.2--urandom_range_1.sv:4::tags: 18.13.2 uvm
  18.13.2--urandom_range_2.sv:4::tags: 18.13.2 uvm
  18.13.2--urandom_range_3.sv:4::tags: 18.13.2 uvm
  18.13.3--srandom_0.sv:4::tags: 18.13.3 uvm
  18.13.4--get_randstate_0.sv:4::tags: 18.13.4 uvm
  18.13.5--set_randstate_0.sv:4::tags: 18.13.5 uvm
  18.14--random-stability_0.sv:4::tags: 18.14 uvm
  18.14--random-stability_1.sv:4::tags: 18.14 uvm
  18.14--random-stability_2.sv:4::tags: 18.14 uvm
  18.14--random-stability_3.sv:4::tags: 18.14 uvm
  18.14.2--thread-stability_0.sv:4::tags: 18.14.2 uvm
  18.14.2--thread-stability_1.sv:4::tags: 18.14.2 uvm
  18.14.3--object-stability_0.sv:4::tags: 18.14.3 uvm
  18.14.3--object-stability_1.sv:4::tags: 18.14.3 uvm
  18.15--manually-seeding-randomize_1.sv:4::tags: 18.15 uvm
  18.16--random-weighted-case-randcase_1.sv:4::tags: 18.16 uvm
  18.16--random-weighted-case-randcase_3.sv:4::tags: 18.16 uvm
  18.17--random-sequence-generation-randsequence_1.sv:4::tags: 18.17 uvm
  18.17--random-sequence-generation-randsequence_3.sv:4::tags: 18.17 uvm
  18.17.1--random-production-weights_1.sv:4::tags: 18.17.1 uvm
  18.17.2--if-else-production-statements_1.sv:4::tags: 18.17.2 uvm
  18.17.2--if-else-production-statements_3.sv:4::tags: 18.17.2 uvm
  18.17.3--case-production-statements_1.sv:4::tags: 18.17.3 uvm
  18.17.4--repeat-production-statements_1.sv:4::tags: 18.17.4 uvm
  18.17.5--interleaving-productions-rand-join_1.sv:4::tags: 18.17.5 uvm
  18.17.5--interleaving-productions-rand-join_3.sv:4::tags: 18.17.5 uvm
  18.17.6--aborting-productions-break-and-return_1.sv:4::tags: 18.17.6 uvm
  18.17.6--aborting-productions-break-and-return_3.sv:4::tags: 18.17.6 uvm
  18.17.7--value-passing-between-productions_1.sv:4::tags: 18.17.7 uvm

```

tgorochowik, 2020-11-20:
I see how this can be a nuisance for you and I agree that there is reason to somehow change that.
From the options you listed I think number 2 is the most reasonable, however I would prefer to keep the chapter tags too.

Maybe we could just create UVM-specific tags and treat them separately?
We could change the tags from
`tags: 18.13.1 uvm`
to
`tags: uvm uvm-18.13.1`

This way we don't lose anything and it still solves the issue for tool developers not interested in uvm.

CC @hzeller

wsnyder, 2020-11-20:
I like renaming the tag, would it be better

    tags: uvm uvm-18.13.1

or just

    tags: uvm-18.13.1

Generally when reviewing the dashboard I prefer a single tag as then don't look at a failing test twice ;)

tgorochowik, 2020-11-20:
I think that right now the `uvm` tag is a kind of "catch-all" tag, it doesn't have any specific tests. So for now I'd keep both tags, you should be able to just skip browsing the `uvm` tag then. I think that it should be ok to remove it, but If we decide to do that we should do it globally (and only keep the smaller `uvm-*` tags).

And I actually just noticed that for chapter 16 we use just the `uvm` and the `uvm-assertions` tags - there is no information about a particular sub-chapter. So I guess changing everything from chapter 18 to `uvm` and `uvm-random` (to keep the name short) would make sense too as it would at least be consistent with chapter 16.

hzeller, 2020-11-20:
Collapsing most of chapter 18 that import UVM is somewhat defeating the purpose of labeling them with the necessary chapters though I understand that it looks 'nicer' in some contexts.

Could we have something where we ifdef-check for some define `BUILDIN_RANDOM` and only import uvm_pkg::* if that is not there ? Then we could do some define-hacks for tools that support random but not necessarily uvm ?

tgorochowik, 2020-11-20:
We still have some tests in chapter 18 that do not use uvm. The labels for them should remain unchanged.

@hzeller would rather we used the `uvm-18.X.X` scheme? Would that be acceptable?

hzeller, 2020-11-20:
I think `uvm-18.X.X` that would be nicer

closed 2020-11-26 by tgorochowik

tgorochowik, 2020-11-27:
@hzeller I am not entirely happy with the way the report looks now - it looks kind of polluted with the uvm specific tags - I think we may want to reconsider using the aggregate `uvm-random` tag in this case

PR #1147 (2020-11-26): Remove chapter tags from UVM tests
Basically a recursive `s/tags: 18.* uvm/tags: uvm uvm-random/g` in `tests/chapter-18`.

Closes: #1146