chipsalliance/sv-tests #1155: Icarus failing various number_test_* checks

caryr, 2020-11-21:
As a two stage compiler Icarus verifies that there is a top level module since that is required for generating running code. I noticed that many of the number_test_* files just have a plain parameter definition which is correct for just parsing, but Icarus does both parsing and elaboration. To make these files compatible with Icarus can you add the following to the failing tests listed below.
```
`ifdef __ICARUS__
module dummy;
endmodule
`endif
```
This should allow the tests listed below to pass:
- number_test_0-16
- number_test_24-36
- number_test_42-53
- number_test_59-62
- number_test_64-72

tgorochowik, 2020-11-23:
@caryr thanks for the report! It should be ok to add the module definition to those tests even without the ifdef.

Would you be willing to create a PR for that?

MikePopoloski, 2020-11-23:
Not that I disagree with improving the tests, but most tools are able to just print a warning if there is no top module and still compile and check the code. Might be easy enough to modify Icarus to do the same?

caryr, 2020-11-23:
Yes, I can look at adding parse only support to Icarus. As a minimum it would require a parse only flag since we want to flag a missing module as an error when actually compiling for simulation.

I can look at creating a PR, are there directions for downloading sv-tests and testing using just a single external simulator/test suite? Having that would make it easier for the individual tools providers to test/improve their support.

We are also reworking ivtest so it better supports running all tests as SystemVerilog using an appropriate `begin/end_keywords pair for many of the failing files. This should improve the passing percentage as well as remove a bunch of the black listed files. There are still some others that will require further discussions, but lets get ride of the obvious ones first.

tgorochowik, 2020-11-23:
> (...) are there directions for downloading sv-tests and testing using just a single external simulator/test suite? Having that would make it easier for the individual tools providers to test/improve their support.

By default sv-tests look for tool executables in PATH, if you do have some extra tools installed, that you just don't want to test feel free to temporarily remove their wrappers from `tools/runners/`, that should do the trick.

I guess you could also modify the Makefile to allow generating the report for just a single tool. Adding an option to run all the tests for a single tool should be straightforward as you could just create a new target in the generator - the part with generating the actual report without trying to rebuild everything may need some more effort. Feel free to send a patch if you come up with something useful :)

caryr, 2020-11-26:
It looks like the code added in #1164 could also resolve this when the tests listed above are updated to use the new functionality.

tgorochowik, 2020-11-27:
Unfortunately we won't be able to use it for these tests as #1164 only changes the [path_generator](https://github.com/SymbiFlow/sv-tests/blob/94f536e54110e1453d2502e1e00be6c41fa2da27/generators/path_generator). The tests you mention are used directly and won't be affected by the generator changes.

caryr, 2020-12-02:
Given the BW I have available cloning this repo is not something that is currently feasible for me. Ideally there would be a way to download just the basic test suite and then pull the included test suites and other pieces separately. I realize this is likely too much work to satisfy one person with a slow connection. I will not be able to create a PR for this.

tgorochowik, 2020-12-02:
@caryr note that you can create the PR even from a shallow clone:
```
git clone git@github.com:caryr/sv-tests.git --depth 1
```
(assuming you forked `sv-tests`)

This is as tiny as it gets :)

You can then selectively clone third-party modules the same way (e.g. the third-party tests)

Note: we do have to fix the size of the repo anyway (see #1179), but a shallow clone will always be the tiniest package.

caryr, 2020-12-02:
Thanks. I will try this later.

caryr, 2020-12-15:
I think test should be designed so that they will eventually elaborate and ideally generate a pass/fail result. Limiting things to just parsing seems short sighted for most tools. I am in the process of updating the tests to have a top level module and will submit that once completed.

caryr, 2020-12-15:
Fixed in #1210

closed 2020-12-15 by caryr