chipsalliance/sv-tests #1160: Many basejumpSTL tests are bogus

MikePopoloski, 2020-11-25:
Many of the basejump tests don't work correctly as standalone tests. They reference modules from other files that aren't included in the tests, for example. Most annoyingly, they default all module parameters to "inv" which makes them look like valid top-level modules. All of the tools that correctly follow the SystemVerilog rules for finding top modules will instantiate these and then get stuck in generate loops for millions of iterations. If you look at the top 10 slowest tests for Icarus, Verilator, and slang, they are all basejump STL tests:
```
BaseJumpSTL_bsg_channel_tunnel_wormhole
BaseJumpSTL_bsg_nonsynth_dramsim3
BaseJumpSTL_bsg_mem_multiport
BaseJumpSTL_bsg_nonsynth_ramulator_hbm
BaseJumpSTL_bsg_cache
BaseJumpSTL_bsg_crossbar_control_basic_o_by_i
BaseJumpSTL_bsg_fsb
BaseJumpSTL_bsg_fifo_shift_datapath
BaseJumpSTL_bsg_lru_pseudo_tree_backup
BaseJumpSTL_bsg_cache_to_axi
```

Perhaps this "test suite" should just be removed since it doesn't actually function as a standalone test suite? Alternatively, it seems like you may need a new mode to mark these as "parse-only" or something, to indicate that tools should definitely not try to do type checking or elaboration for them because it's guaranteed to fail. The existing "parse" tag conflates the modes too much to be useful for this I think.

taylor-bsg, 2020-11-25:
The commercial tools all handle BaseJump STL okay. BaseJump STL is one of
the more advanced synthesizeable open source corpuses of SV code out there
today.

Not sure what you mean by “inv” causing modules to look like top level
modules? What rules are you referring to specifically?

The use of inv technique is use to indicate that the module is “abstract”
and should not be instantiated without specifying those parameters. (If you
have a better way of doing this in SV let me know!) No BaseJump code
instantiates those modules without specifying the parameters. I am not sure
if these tools should be blindly elaborating uninstantiated modules.

M

MikePopoloski, 2020-11-25:
Commercial tools have exactly the same problem. If I pass one of above mentioned tests into VCS 2020.03, it hangs for many many seconds. The problem, as I explained, is that if you say "please elaborate this module" and the module has parameters set to "inv" they are elaborated as huge integer numbers and generate loops that reference them can run for a very long time. The rule in SystemVerilog, according to the LRM, is that any module that is not referenced elsewhere by an instantiation and has no parameters or *all parameters have default values* is considered a top-level module and required to be automatically instantiated. The correct way to require that a parameter be provided a value is to *not give it a default at all*.

Here's the simplest possible example: https://www.edaplayground.com/x/BaZt if you run that with VCS, it will run for nearly 20 seconds!

You may argue that nobody should try to include such a module in their design unless they desire to use it, but all of the tests imported by the SymbiFlow test suite *do* this, so... my original argument was that they should not. They are not meant to be tested standalone.

taylor-bsg, 2020-11-25:
> *EDA Playground <https://www.edaplayground.com/x/BaZt>*
> Edit, save, simulate, synthesize SystemVerilog, Verilog, VHDL and other
> HDLs from your web browser.

Okay, how about this:

https://www.edaplayground.com/x/s32A

M

taylor-bsg, 2020-11-25:
Per this:

 Top-level modules are modules that are included in the SystemVerilog
source text, but do not appear in any module instantiation statement, as
described in 23.3.2. This applies even if the module instantiation appears
in a generate block that is not itself instantiated (see 27.3). A design
shall contain at least one top-level module. A top-level module is
implicitly instantiated once, and its instance name is the same as the
module name. Such an instance is called a top-level instance.

It seems like the issue is that these "virtual" modules are getting
identified as top-level modules. With the macro, we can add a dummy
instantiation so that it is not considered as top-level.

M

MikePopoloski, 2020-11-25:
Yes, that helps. Giving the design an explicit list of top-level modules to instantiate also solves it, but you need at least one so the dummy helps with that. You can get around the need for the macro though; as long as you have the top module, all tools I know of let you specify the list of top-level modules explicitly on the command line, and SymbiFlow has a test tag to specify it.

I think it should also be sufficient to just leave off the default values for the parameters, but ironically the commercial tools all seem to have trouble with that -- maybe someday they'll fix it.

MikePopoloski, 2020-11-25:
The spec says in 6.20.1:
`Also, if no default value is specified for a parameter of a design element, then a tool shall not implicitly instantiate that design element`
so the commercial tools really ought to respect that, but alas from my testing they do not.

taylor-bsg, 2020-11-26:
Thanks Mike. I am going to circulate this with the BaseJump STL developers.

MikePopoloski, 2020-11-26:
Thanks for looking into it. I don't think you will want to add these extra module definitions to every file. It seems more likely that the SymbiFlow tests can have a script that inserts them automatically, since they're only needed if you want to compile a particular file standalone.

tgorochowik, 2020-11-26:
We could probably create a jinja template for that and fill it before generating the testcase. This should be fairly straightforward assuming all the filenames are the same as the modules in them (it looks as though they are).

MikePopoloski, 2020-11-26:
You could also just have a separate file that contains `module top; endmodule` and include that in each basejump test, and then set the top_module tag to "top" so that you don't have to modify the individual files.

tgorochowik, 2020-11-27:
There are some issues with the generator changes, let's keep this open for the time being

closed 2020-12-01 by tgorochowik

tgorochowik, 2020-12-01:
@MikePopoloski there are still many tests that fail with slang after including the fake top module. Is this to be expected?

MikePopoloski, 2020-12-01:
I think they will pass whenever slang is updated to more recent build. I fixed an issue a week or two ago where unknown modules inside uninstantiated modules would still error.

PR #1164 (2020-11-27): Generate a fake topmodule for basejump tests
Fixes #1160 

@MikePopoloski could you please take a look if this solves the issue for you?
PR #1183 (2020-12-01): Include the actual test code in path-generator tests
This is needed to include the fake topmodule code added earlier.

Closes #1160