chipsalliance/sv-tests #1169: How to blacklist hdlconvertor tests?

wsnyder, 2020-11-27:
How do I blacklist illegal hdlconvertor tests?

I don't see a file similar to e.g. generators/ivtest which has a blacklist for Icarus tests

```
hdlconvertor_adder_implicit.sv.log
hdlconvertor_directive_verilogpp.sv.log
hdlconvertor_fifo_rx.sv.log
hdlconvertor_p138.sv.log
hdlconvertor_p174_2.sv.log
hdlconvertor_p176.sv.log
hdlconvertor_p180_2.sv.log
hdlconvertor_p229.sv.log
hdlconvertor_p335.sv.log
hdlconvertor_p342.sv.log
hdlconvertor_p504.sv.log
hdlconvertor_p524.sv.log
hdlconvertor_p711.sv.log
hdlconvertor_p713.sv.log
hdlconvertor_p714.sv.log
hdlconvertor_p731.sv.log
hdlconvertor_p732.sv.log
hdlconvertor_p734.sv.log
hdlconvertor_p735.sv.log
hdlconvertor_p736.sv.log
hdlconvertor_p755.sv.log
hdlconvertor_p77.sv.log
hdlconvertor_p782.sv.log
hdlconvertor_p793.sv.log
hdlconvertor_p794.sv.log
hdlconvertor_p795.sv.log
hdlconvertor_p802.sv.log
hdlconvertor_p825.sv.log
hdlconvertor_p836.sv.log

```

tgorochowik, 2020-11-30:
We use common generators where it is possible and just provide json configs for them.

In this case it is: https://github.com/SymbiFlow/sv-tests/blob/master/conf/generators/meta-path/hdlconvertor.json

wsnyder, 2020-11-30:
Thanks - will make pull request later so closing.

closed 2020-11-30 by wsnyder