chipsalliance/sv-tests #1179: Fresh clone sv-tests takes 4-5 hours -  is this common?

svenka3, 2020-11-28:
Hello,
  I am trying a fresh git clone of sv-tests on a Linux laptop (not very high-end, but does run EDA tools and is connected to highspeed Ethernet (Tried WiFi first, then moved to Ethernet). The process seems to take multiple hours > 4 hours. Is this common? I am sure it is first-time-only issue, later should be faster, but just wanted to check with others. Attaching a screenshot.

Any comments appreciated.

Thanks
Srini

[image]

mithro, 2020-11-30:
@svenka3 - I would not expect that...

mithro, 2020-11-30:
Oh, I see what is going on.

mithro, 2020-11-30:
```
tansell:~/tmp$ time git clone https://github.com/SymbiFlow/sv-tests.git --single-branch --branch master
Cloning into 'sv-tests'...
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 11663 (delta 0), reused 1 (delta 0), pack-reused 11662
Receiving objects: 100% (11663/11663), 5.08 MiB | 12.21 MiB/s, done.
Resolving deltas: 100% (6744/6744), done.

real    0m1.559s
user    0m0.603s
sys     0m0.266s
```

The issue is the `gh-pages` branch is *massive*, like >1gb.

mithro, 2020-11-30:
I think we probably need to move the gh-pages into its own repository?

mithro, 2020-11-30:
@hzeller - What are your thoughts?

mithro, 2020-11-30:
I tried to see if there is a way to make `git  clone` to ignore the `gh-pages` branch by default when cloning, but that doesn't seem to be supported.

I don't think we want to require people to run a special `git clone` command, as people won't notice they have to.

I think the steps would be to create a `sv-tests-output` repository, push the `gh-pages` to that repository and update the scripts to push there, remove the `gh-pages` branch from this repo and finally create a *new* `gh-pages` branch in this repository which redirects to the new https://symbiflow.github.io/sv-tests-output location?

mithro, 2020-12-30:
I have done the following;
 * created an [sv-tests-results repository](https://github.com/SymbiFlow/sv-tests-results).
 * push the current contents of sv-tests-results's gh-pages branch to the gh-pages result there.
 * setup github pages on the sv-tests-results repo
 * created a gh-pages-redirect branch in this repo which redirects from https://symbiflow.github.io/sv-tests to https://symbiflow.github.io/sv-tests-results
 * made the gh-pages-redir branch the github pages branch for this repository

Final things to do are;
 * Update CI to push to the right locations
 * Remove the gh-pages branch from this repository

mithro, 2020-12-31:
After merging #1242 and removing the gh-pages branch from this repository. This bug should hopefully be fixed.

PR #1242 (2021-01-01): kokoro: Push to SymbiFlow/sv-tests-results repository.
As the HTML results being in the gh-pages branch end up with a rather
large and slow to clone repository, the results have been moved to their
own repository.

This makes the CI push to that repository instead.

Fixes #1179.
commit 94744f78 (2021-01-01): kokoro: Push to SymbiFlow/sv-tests-results repository.

As the HTML results being in the gh-pages branch end up with a rather
large and slow to clone repository, the results have been moved to their
own repository.

This makes the CI push to that repository instead.

Fixes #1179.
closed 2021-01-01 by mithro