chipsalliance/sv-tests #1181: dependabot hasn't updated slang in a few weeks

MikePopoloski, 2020-11-28:
I don't know much about how dependabot works -- any idea what might be wrong here? I don't see any opened or closed pull requests from it for slang for several weeks.

tgorochowik, 2020-11-30:
That is right, it seems to be also correct for Verible. There is something wrong with dependabot, we need to look into it.

Note that the tool submodules are only used for local builds - the CI (and the report) uses conda packages.

MikePopoloski, 2020-11-30:
I see that the report version of slang is indeed slightly different from the last dependabot PR, but the report version is also out of date as well. The report version was last updated Nov 16th, and I was under the impression it's supposed to update weekly?

tgorochowik, 2020-11-30:
The packages that are used here in the CI are the packages from https://github.com/symbiflow/conda-packages

So these weren't really weekly updates - the CI was triggered there each time a new commit was pushed to that repo.

Currently the issue is that the conda package repository still uses Travis which is practically dead. We do plan to migrate it to gh actions, we have it in our queue but we still did not get around to doing this.

Assuming the port to gh actions is done, github allows specifying cron jobs on when to fire these actions and then weekly updates (in addition to repo updates) will probably be a good idea.

MikePopoloski, 2020-11-30:
Ah, I see, thanks for the explanation.

MikePopoloski, 2021-01-03:
Any idea when the conda packages will start being updated again? It would be nice to have changes and fixes start showing up in the CI report again.

tgorochowik, 2021-01-04:
@MikePopoloski we have some PRs in place to prepare the conda infrastructure with gh pages on litex hub: https://github.com/litex-hub/litex-conda-eda/pull/64

And most of the packages are already ported too but temporarily sit on a different branch, e.g: https://github.com/antmicro/litex-conda-eda/tree/pcza-move-to-gha/sim/slang

Once this is all done we will just have to update sv tests to use the new channel (litex hub). If nothing unexpected pops up it should be done quite soon.

caryr, 2021-01-10:
It looks like the repo was updated with the latest slang a couple days ago. Not sure how this translates into what is used for generating the dashboard.

MikePopoloski, 2021-01-10:
Yeah, not sure either, but glad to see it updated.

closed 2021-01-10 by MikePopoloski