chipsalliance/sv-tests #1186: Many tests from ivtest can now be removed from the blacklist.

caryr, 2020-12-02:
I have updated the ivtest regressions to support being run exclusively using SystemVerilog instead of using the generation we provide in the regression list. I have added appropriate `begin_keywords/`end_keywords and a couple other minor tweaks to allow the test to work using iverilog with the generation overridden to be -g2012. You will still need to keep the AMS files blacklisted and it looks like a few others need to be tweaked to get them to pass for other parsers. I wanted to take care of this first to get down to just the ones with real portability issues.

You will need to sync to the latest ivtest and I can provide help dealing with any issues you have, but like I said in the other issue I was planning to look at the sv-tests repo is too large for me to download to my development machine since my network connection is too slow for something this large. I cannot let this run for days since I need to use the BW I have for real work.

If you sync to the latest iverilog then you should be able to test this using your scripts and the latest ivtest and once that is clean for iverilog see what the other parsers do. I needed to fix an issue in iverilog yesterday to make the tests clean so make sure you sync that as well as ivtest

tgorochowik, 2020-12-02:
Thanks for the update!

Would you mind trying with the shallow clone method I described in #1155 ? The repo footprint is then considerably smaller:

```
$ git clone git@github.com:symbiflow/sv-tests.git --depth 1
Cloning into 'sv-tests'...
remote: Enumerating objects: 1596, done.
remote: Counting objects: 100% (1596/1596), done.
remote: Compressing objects: 100% (1345/1345), done.
remote: Total 1596 (delta 389), reused 672 (delta 236), pack-reused 0
Receiving objects: 100% (1596/1596), 793.71 KiB | 1.72 MiB/s, done.
Resolving deltas: 100% (389/389), done.
$ cd sv-tests
$ du . -csh
9.7M    .
9.7M    total
```

Is that something you can work with?

caryr, 2020-12-02:
That's certainly a manageable size. It's not like I have dial-up ;). I will try this and see if I can at least get ivtest working cleanly for iverilog. Does the testsuite handle missing sub-tests gracefully or is that the first thing that needs to be added?

caryr, 2020-12-15:
It appears that the method described above and in #1155  is not coping enough of the files. For example the generators directory does not have a ivtest directory.

I also tried to run make and that just failed so I have been testing all my changes directly using iverilog and then pushing and using CI to check it works for everything else. That works for what I am doing now, but I cannot work on this report until I can get at the missing files.

tgorochowik, 2020-12-15:
Thank  you for working on this!

As mentioned in #1155 the submodules have to be fetched separately, you can do this globally, or selectively, for example, for ivtests:
```
$ git submodule init
$ git submodule update --depth 1 third_party/tests/ivtest/ 
```

I hope it helps!

caryr, 2020-12-15:
I realized I needed that for things in the tests directory, but was not expecting that for files in the generators directory. I'll clone ivtest as you describe above and see if that releases some dependency for the other files.

Thanks for the above commands. I had not looked them up yet.

caryr, 2020-12-15:
FYI no idea how I missed this, but it looks like the generators/ivtest file was being downloaded correctly. I will probably have a few more questions later. I have submitted a fix that should allow the latest ivtest to be pulled in and once that is done I can start working on this and other cleanup related to ivtest.

Thank you for your help so far!

caryr, 2020-12-30:
The last of the tests have been reworked in #1239

closed 2020-12-30 by caryr