chipsalliance/sv-tests #1208: Verilator version on dashboard appears wrong

wsnyder, 2020-12-13:
The version on the dashboard for Verilator seems very old. This doesn't seem to correspond with either the pass/fails which suggest more recent (ignoring #1207 false fails),  or version in the third_party directory which is much more recent. Is there another version mis-installed on the dashboard generation machines?

[image]

tgorochowik, 2020-12-14:
The package that is used is downloaded from the anaconda repository. It looks as though we are not downloading the latest version for some reason:
https://anaconda.org/SymbiFlow/verilator/files?sort=ndownloads&sort_order=desc

caryr, 2020-12-17:
iverilog also looks to be outdated. Though not as bad as verilator. I'm also not sure about the file names they show 10, but V10 was replaced with V11 stable a while ago. I really think you should be tracking devel since that is where all the new functionality appears. Our general rule is only bug fixes go into stable which likely makes it almost LTS except we don't guarantee support for a given time. The goal for stable is stability :-). We do unstable development in branches and only integrate into master when the test suite is clean.

tgorochowik, 2020-12-30:
This should be fixed by #1240

caryr, 2021-01-01:
I'm not certain how often the dashboard is updated, but based on test failures it looks like Icarus is still out of date. I did update both it and ivtest earlier this week in the repo and ivtest does reflect the latest changes, but not iverilog. Also the anaconda link/version for Icarus looks totally wonky.

wsnyder, 2021-01-05:
Thanks for fixing, just a note I haven't closed this, as I have yet to see the updated version appear on the dashboard (which appears stuck for other reasons).

wsnyder, 2021-01-10:
On today's dashboard version is good, thanks.

closed 2021-01-10 by wsnyder